OctoPrint/OctoPrint #2094: Uploads file dose not support the chinese file name, although I have modified the file according to the issue  '# 1118'.

EvanFei, 2017-08-24:
If there is a file in chinese name, it always show the error "UnicodeDecodeError: 'ascii' codec can't decode byte 0xb2 in position 0: ordinal not in range(128)."
No matter how to transformed the file-coding, it's not working.
Could you please help me to fix the problem?Thank you very much.

foosel, 2017-08-25:
Hi @EvanFei. Could you provide a file with which to test this and also your `octoprint.log` file please? Thanks!

EvanFei, 2017-08-25:
[octoprint.log&gcodefile.zip](https://github.com/foosel/OctoPrint/files/1251319/octoprint.log.gcodefile.zip)

Hi @foosel .I compressed them in [octoprint.log&gcodefile.zip].Thanks for you help.

foosel, 2017-08-25:
It looks like this log file doesn't contain the error at all. Could you please reproduce the error, then provide a log file that contains the logged stack trace?

Also, your log indicates something's up with your OctoPrint installation as it reports "0.000" as its version. The installation location hints at this being version 1.2.13, which was released on June 16th **2016**, so way more than a year ago. The current version is 1.3.4. Please update to the current version and test if you can even reproduce the issue there.

I can't reproduce any problem with the file you provided under 1.3.5rc2, it uploads just fine without any errors (note that its name does get converted to ASCII characters, for internal reasons, but that does work just fine):

[image]

EvanFei, 2017-08-25:
[octoprint.log.zip](https://github.com/foosel/OctoPrint/files/1251469/octoprint.log.zip)
My OctoPrint installation is "1.2.13", I wanna modify in this version to support chinese file name, is that possible?

foosel, 2017-10-17:
Starting with version 1.3.6, uploaded files will persist their full UTF-8 file name in `.metadata.yaml` to be available alongside their ASCII only representation on disk:

[image]

As can be seen, that also allows stuff like emoji in file and folder names.

Already available on `maintenance` and `devel`.

EvanFei, 2017-10-19:
Hi, @foosel .Could you please give me a guide to show how to realize this function.I'm confused.Thanks.

foosel, 2017-10-19:
@EvanFei Wait until 1.3.6 (or 1.3.6rc1 for that matter) gets released, or switch to the `maintenance` branch. See [the FAQ for how to do this with OctoPi](https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-switch-the-branch-of-the-octoprint-installation-on-my-octopi-image).

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel