OctoPrint/OctoPrint #2097: [Brainstorming] Introducing gcode comments to prevent parsing the whole file

malnvenshorn, 2017-08-26:
After uploading a ~30MB gcode file to OctoPrint it took quite some time on my little Raspberry Pi B+ to parse that file. So I thought it would be great to make it unnecessary to parse the whole file. If the information (used filament, estimated print time, dimensions of the model) are present at the beginning of the file the gcode interpreter could use these and stop going further.

ntoff, 2017-08-27:
and how do you plan on getting that information into the gcode files in the first place? not every slicer puts it there, and the ones that do, don't all put it in the same place, and if you're still crazy enough (i.e. me) to even consider using skeinforge, it puts that info in a separate file along side the gcode file.

malnvenshorn, 2017-08-27:
I'm talking about OctoPrint specific comments. Using the comments provided by various slicers is not feasible. Not every slicer produces the same comments or put them in the same place, if they create them at all.

Those comments could look like
```
; octoprint_filament_length = 701.9
[...]
```
and could be created by running a script after the slicer has finished. Basically the code already exists  and if the slicer supports post processing scripts (e.g. slic3r) this can be easily integrated into the workflow.

Maybe it is not a feature for everyone, but if OctoPrint doesn't  find those comments it will parse the file as usual. So nothing will change for those people, but it will speed things up a lot for people like me.

ntoff, 2017-08-27:
ok, I guess that could work, octoprint hasn't bothered parsing any of my gcode for time estimation in forever, I never get print estimates anymore (until I hit print) so if there was a way to get time estimates out of slic3r that *don't* involve me uploading to gcode.ws then I'm all for that.

Andrei-Pozolotin, 2017-09-04:
another relevant performance issue: 
* why load 30MB gcode into client (and crash the browser)? 
* why not stream on demand (+/- 100 lines in context)?

jneilliii, 2017-09-04:
This is actually possible via a plugin.  You could use the example on the [octoprint-filemanager-preprocessor}(http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-preprocessor) documentation page and just take out the check for ending in "_strip" and process all gcode files on upload.

jneilliii, 2017-09-04:
Not sure if completely possible to set the meta data of the analyzed gcode (ie estimated time, etc.) but there are available plugin hooks to preprocess the uploaded files.  Documentation can be found over on the [octoprint-filemanager-preprocessor](http://docs.octoprint.org/en/master/plugins/hooks.html#octoprint-filemanager-preprocessor) section.

malnvenshorn, 2017-09-05:
@jneilliii Good thought, but how do you prevent that the file is nevertheless added to the analysis queue?

malnvenshorn, 2017-11-06:
At the moment I always have to replace the _gcodeInterpreter.py_. It would be great if I could provide the metadata via plugin.

malnvenshorn, 2017-11-09:
Made a plugin, therefore closed.

closed 2017-11-09 by malnvenshorn