OctoPrint/OctoPrint #2098: Webinterface extremly slow

michidk, 2017-08-28:
#### What were you doing?

Installed Octoprint on my Raspberry Pi 3 (Raspbian). Worked fine before.
But a few days ago I did the following: 
I didn't install my Octoprint installation into a virtualenv, I wanted to fix this, so I did the following steps:
- remove Octoprint folder
- uninstalled all python things (python2, python3, pip)
- `apt-get install python2.7`
- `easy_install pip`
- `easy_install virtualenv`
- `easy_install virtualenvwrapper`
- `git clone https://github.com/foosel/OctoPrint.git`
- `cd OctoPrint`
- `virtualenv venv`
- `./venv/bin/pip install pip --upgrade`
- `./venv/bin/pip install -r requirements.txt`
- `./venv/bin/python setup.py install`
- `sudo usermod -a -G tty pi`
- `sudo usermod -a -G dialout pi`
- .`/venv/bin/octoprint serve`
- `mkdir ~/.octoprint`

Note: my OctoPrint folder is located in ~/run/

Also tested with maintenance branch.

#### What did you expect to happen?

The Octoprint webinterface should appear.

#### What happened instead?

Webpages loads about 10 minutes, after that I see a page without design and icons slowly appearing. Tabs changing and every other action takes about 3-10 minutes.

#### Did the same happen when running OctoPrint in safe mode?

Yep, same with `octoprint serve --safe`

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (master branch)
and
OctoPrint 1.3.6.dev37+gb161a1f (maintenance branch)

#### Operating System running OctoPrint

Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie

#### Printer model & used firmware incl. version

Issue has probably nothing to do with printer. Tested both: Printer on and off: Same result. Prusa Anet A8.

#### Browser and Version of Browser, Operating System running Browser

Tested with Chrome, Edge and Firefox on Windows 10

#### Link to octoprint.log

https://pastebin.com/rnEdbdgL

#### Link to contents of terminal tab or serial.log

Can't access settings or terminal tab... Waited 30 minutes for it to load.

#### Link to contents of Javascript console in the browser

```GET http://192.168.10.22:5000/static/js/lib/less.min.js net::ERR_CONNECTION_RESET
VM208:1 This page includes a password or credit card input in a non-secure context. A warning has been added to the URL bar. For more information, see https://goo.gl/zmWq3m.
VM208:1 GET http://192.168.10.22:5000/ net::ERR_CONNECTION_RESET```

#### Screenshot(s)/video(s) showing the problem:

https://i.imgur.com/dUB10jU.png

I have read the FAQ.

foosel, 2017-09-28:
The connection getting reset looks more like a network issue than anything, which is nothing I can do anything about. Did anything change recently with regards to your LAN setup, e.g. did you move the router or the Pi that might explain a slow connection?

michidk, 2017-09-30:
I didn't change anything regarding the network. The Pi is directly connected to my network switch, which is connected to my pc. Other webservices work fine. 
But I changed my SD card to one with more storage, but probably less speed. I benchmarked the sd card using dd: write 17.3MB/s; read 22.7MB/s .
This should be enough for OctoPrint running smoothly, right?

foosel, 2017-10-02:
Yes, it should, especially since the issues you are seeing (connection reset) I would normally not associate with SD card speed. But it still might be interesting to see if the behaviour persists if you swap back to the old one. Is that possible?

Also, when it happens, can you still ping the Pi and if so what round trip times are you seeing? Can you still login via SSH? Does the same happen when running OctoPi?

michidk, 2017-10-02:
I'm a bit stunned now...

I wanted to test the ping, pulled a new version of Octoprint, installed it and started, and it just worked. 
I don't know if the new version fixed it or just waiting a month... I did't change any configuration. Network is still the same.

I hope it keeps working.

Thank you for your support!

closed 2017-10-02 by michidk