OctoPrint/OctoPrint #2104: Allow git+https URIs for plugins

JohnMHarrisJr, 2017-08-31:
OctoPrint currently only allows http or https URLs to be added as plugins. This is an issue for some servers, such as cgit installations. This should be as simple as allowing URLs prefixed with "git+", which pip has special handling for.

foosel, 2017-09-28:
> OctoPrint currently only allows http or https URLs to be added as plugins

What location are you referring to here? This?

[image]

JohnMHarrisJr, 2017-09-28:
Yes, that's the portion I'm referring to. Sorry for the lack of info.

foosel, 2017-10-02:
Implemented in the above commit, pushed to `maintenance`, soon `devel`, will be part of 1.3.6.

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel