OctoPrint/OctoPrint #2105: how print only with sd?

EnricoSx, 2017-09-01:
hello, i have a gcode in sd, i want print without risk of usb disconnect, how can i do? thanks.

nophead, 2017-09-01:
You need a control board that has a jumper to disable reset on USB
disconnect.

EnricoSx, 2017-09-06:
**HOW CAN I USE THIS COMMAND?**

My SD prints get interrupted when I connect to my printer from OctoPrint!

Your printer has a controller board that resets on a new connection via the serial interface. A lot of them if not the majority do. There's nothing OctoPrint can do about that, it's a "feature" of your printer's hardware.

Since OctoPrint can't query the printer status without connecting to it, it also can't warn you about that before you hit connect (and displaying a warning every time you hit connect would be incredibly annoying to the majority of users who don't run SD prints outside of OctoPrint's control).

Nonetheless, the following works for some boards:

stty -F /dev/ttyACM0 -hupcl
which will disable "the usual" USB -> Serial -> CTS/DTR, etc -> RESET action

ManuelPeiro, 2017-09-10:
hi,

i have the similar problem.
Were i insert the code?

stty -F /dev/ttyACM0 -hupcl

No?

EnricoSx, 2017-09-10:
I don't know :(

closed 2017-09-16 by GitIssueBot