OctoPrint/OctoPrint #2106: [Request] Change the order of tabs

Boogy62, 2017-09-02:
Hi ! Is there a way to change the order of tabs (Temperature, Control...) ? When many plugins are installed, main tabs are hidden...

Thanks

ntoff, 2017-09-03:
This is a bug tracker, not a support line, in future please use the google + group https://plus.google.com/communities/102771308349328485741 or the mailing list https://groups.google.com/forum/#!forum/octoprint for help / support.

There is no GUI way to alter the tab order, although you can edit config.yaml to reorder the tabs. http://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance

Boogy62, 2017-09-03:
> master/configuration/config_yaml.html#appearance

Thanks, I didn't know where was the right place to post.

ntoff, 2017-09-03:
No problem, I hope I didn't come across as being angry or anything, did that configuration documentation answer your question sufficiently? Do you need help figuring out how to find the names of the tabs?

Boogy62, 2017-09-03:
No it's okay I found everything easily.

Thanks

foosel, 2017-09-26:
Closing since this seems to have been solved.

closed 2017-09-26 by foosel

jneilliii, 2018-01-20:
New improved GUI way via plugin.

https://github.com/jneilliii/OctoPrint-TabOrder