OctoPrint/OctoPrint #2107: Automatic upgrade does not work anymore after manual rollback

pckeller29, 2017-09-05:
READ THE FOLLOWING FIRST:
#### What were you doing?

A few weeks ago I updated Octoprint to 1.3.5 RC1 which caused Octoprint not starting anymore because of a known issue.
Then, using the FAQ, I successfully rolled back to 1.3.4.
But after this manual operation, normal automatic upgrades don't work anymore.

I go to settings menu, Software update (I'm set to RC version) and do a Check for updates (running 1.3.5RC2 as current version)).
I then have a popup telling me there is a new 1.3.5RC3 version. I choose Upgrade, confirm and the upgrade process starts. After 1 or 2 minutes, Octoprint asks me to reload and to reconnect which runs fine.
I then check Octoprint release which is still 1.3.5RC2.

I also tried manually using this command:

`/home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --python="/home/pi/oprint/bin/python" "/home/pi/OctoPrint" "1.3.5rc3"`
Please have a look at pastebin link.

#### What did you expect to happen?
Successfull upgrade from 1.3.5RC2 to 1.3.5RC3

#### What happened instead?
No upgrade occurred despite a normal upgrade process as far as I can see.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5rc2 (master branch)

#### Operating System running OctoPrint

Octopi 0.12.0

#### Printer model & used firmware incl. version
Dagoma Discovery 200
Marlin 1.0RC3

#### Browser and Version of Browser, Operating System running Browser

Firefox 55.0.2
Linux Kubuntu 16.04

#### Link to octoprint.log

Octoprint.log:
https://pastebin.com/Pmp6jCv6
Plugin software update log:
https://pastebin.com/GJTuTMRb

#### Link to contents of terminal tab or serial.log

N/A

#### Link to contents of Javascript console in the browser

N/A

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

foosel, 2017-09-05:
> `/home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.2-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --python="/home/pi/oprint/bin/python" "/home/pi/OctoPrint" "1.3.5rc3"`

It looks like it's using an old version of the update script which doesn't yet know how to handle RCs. The question is why, the path of the script to use should usually be dynamically decided based on the currently running version.

Could you check if your `config.yaml` maybe contains an entry for `plugins.softwareupdate.octoprint.script` (or similar - not 100% sure right now how exactly the entry should be called but it should be fairly obvious of it's there) that points to that version specific update script?

If so, please test of removing it and restarting does solve the problem.

It would also explain why your instance is apparently running off the master branch instead of rc/maintenance, even though it's an RC.

If that's not it, this will have to wait a bit since I'm currently down sick.

pckeller29, 2017-09-05:
Well, I removed "update_script" from configuration.yaml, restarted octoprint, tried an automatic update and it worked fine. I also switched to OctoPrint 1.3.5rc3 (rc/maintenance branch).
Thank you very much (and a special thank for handling my request while sick).

closed 2017-09-05 by pckeller29