OctoPrint/OctoPrint #2108: [Request] Passwords sent in plain text?

maurbeck, 2017-09-08:
I was logging in and by some fluke or by hitting stop or something the address bar showed that I was sending my log in information through plain text to my Octopi server.

It was something like octopi.local/?username=....

This is insecure, no?

foosel, 2017-09-08:
Sounds like the same thing as #1911. As explained there OctoPrint's login API does not accept GET parameters, and the stock UI will also not send those.

So whatever you are seeing must be caused either by a wonky plugin (please test of you can reproduce in safe mode) or some very specific steps.

If you could provide any way to reproduce this, it might be possible to fix.

Since OctoPrint doesn't do SSL termination on its server (on OctoPi that's done by haproxy) and the only option is a self signed certificate anyhow (internal server, no way to generate certificates that will be accepted by a browser out of the box) your credentials will out of the box go over http, unless you access the interface through https (terminated either being OctoPi's haproxy whose certificate you'll have to accept manually or another reverse proxy of your choosing). This is not ideal but the way https workd simply doesn't allow anything better than this at the moment.

maurbeck, 2017-09-08:
So over HTTP you can only send the passwords as plaintext in the payload?

Also, should I expect to be able to view my webcam, see all the files on the pi, and download timelapses when I'm logged out?

Thank you.

maurbeck, 2017-09-08:
[image]

I found that when I quickly log in or out back and forth while the page keeps trying to load in between it has the UN and PW shown in the address bar, shown above.

[image]

This is the image of the webcam streaming while logged out.

foosel, 2017-09-08:
Since I cannot share a secret with a client to encode a password via HTTP and still expect it to stay a secret  and it doesn't make much sense (not is it recommended) to establish a home cooked full blown asymmetric encryption mechanism with perfect forward secrecy and so on when that already exists in the form of HTTPS: yes, I don't see how to have the client send the password to the server in something that doesn't translate to plain text via HTTP, but I'm open to suggestions. If you want your password to be encrypted in transfer, use HTTPS. Already available on OctoPi out of the box (self signed since it's thankfully not possible to get CA certificates for local hosts), otherwise achievable with an additional reverse proxy in front (which is recommended anyhow).

To your other question, yes, this is how it's currently implemented. See #1110 for the ticket on changing that. To fully deny anonymous access, utilize something like basic auth in your reverse proxy for now.

foosel, 2017-09-08:
Also please refer to this FAQ entry: https://github.com/foosel/OctoPrint/wiki/FAQ#why-is-there-no-access-control-for-the-webcam-in-octoprint

maurbeck, 2017-09-08:
Ok, thanks for the explanations.

foosel, 2017-09-08:
> I found that when I quickly log in or out back and forth while the page keeps trying to load in between it has the UN and PW shown in the address bar, shown above.

Would it be possible to get this on video or as aGIF or something? You might want to create a throw away user account for that. I want to make sure I understand your reproduction steps correctly.

Also please test if you can reproduce this behaviour in safe mode. See http://docs.octoprint.org/en/master/features/safemode.html

maurbeck, 2017-09-08:
Essentially, I'm clicking log in, autofilling creds, logging in, and then opening the drop down and logging out. I continue doing that back and forth until the address bar ends up filled as pictured. I log in and out and so forth as soon as the main header appears, which is before the rest of the page completely loads.

I have a print going now and can make a dummy acct and reproduce after that.

foosel, 2017-09-28:
@maurbeck any news on if that's reproducible in safe mode?

foosel, 2017-10-02:
I couldn't reproduce it like you described (either I was not fast enough or my setup too fast in loading the page). But it got me thinking... if you were immediately hitting login before the page was fully loaded, chances were also high that it was not fully initialized and hence the form not properly "redirected" so to speak. I tested this by disabling full page initialization locally and could reproduce. The above commit fixes this by setting a sensible default on the login form ("do not submit anything until fully initialized").

Patch is pushed to `maintenance`, soon `devel`, will be part of 1.3.6.

closed 2017-12-12 by foosel

maurbeck, 2017-12-13:
Great! Sorry that I didn't get a chance to test it like you asked me to earlier.

commit 0f8f4579 (2017-10-02): Add no-op default action to login form

That way people clicking too fast on login before the view models are
bound won't be able to make the form submit username and password as GET
parameter to the base URL.

Closes #2108