OctoPrint/OctoPrint #2111: [Request] - PLEASE Make Scroll Bar Wider, and Maybe Static (do not fade away)

Eddiiie, 2017-09-10:
Hi,
Thank you everyone for creating Octoprint.

As I get older things are getting smaller.  Screen size is increasing but font size is decreasing. lol

The scroll bar on the Files list drives me nuts.  It is SO small and it disappears at the most inconvenient times.   Would it be possible and acceptable to make this scroll bar wider and make it stay?   So many times I miss it and wind up highlighting all the text, reversing the background on the text, pain.

In fact, in general, keep all scroll bars big.  To me, it is worth the space on the screen for easier control.  Who came up with the idea that scroll bars are just too big?   I mean in general... :)

Thank you for your consideration!

ntoff, 2017-09-11:
I find myself using the "Files" section less and less, have you tried the File Manager plugin? http://plugins.octoprint.org/plugins/filemanager/ I don't remember off the top of my head how well the scroll bar works but it's definitely more user friendly than the default tiny little box (especially on my tablet that is too big for touchUI but too small for the default UI).

BillyBlaze, 2017-09-11:
OT: TouchUI will introduce customisation of fonts in the upcoming release

JohnOCFII, 2017-09-11:
> The scroll bar on the Files list drives me nuts. It is SO small and it disappears at the most inconvenient times. Would it be possible and acceptable to make this scroll bar wider and make it stay? [...]

What operating system are you using?  If you are using MacOS, you can change a setting to keep scroll bars always visible.  This is in System Preferences --> General on your Mac.

Eddiiie, 2017-09-12:
This is cool but still need bigger controls.  :D

foosel, 2017-10-17:
Scrollbar is now 10 instead of 5px and does no longer fade out (that it did at all was a bug). Can't go any wider without running into problems with the current layout (and changing that is a whole different undertaking).

Already pushed to `maintenance` and `devel` and will be part of the 1.3.6 release.

ntoff, 2017-10-17:
I now get this on the maintenance branch

>OctoPrint 1.3.6.dev73+gd0aa81c (maintenance branch)

[image]

foosel, 2017-10-17:
Hm, maybe a caching issue? Not seeing that right now on the exact same version. But I had the feeling there's still something off with the cache validation of the static assets.

ntoff, 2017-10-17:
oh ok, sorry, I saw the slimscrolledforeach and thought the new scroll bar was  the reason.

foosel, 2017-10-17:
You and me both on first look, no worries ;) And a fix is incoming.

Eddiiie, 2017-10-17:
FIXED!  THANK YOU!   Yes I am yelling! 
:)

closed 2017-10-17 by Eddiiie