OctoPrint/OctoPrint #2113: TemplateNotFound after trying to use UIPlugin without restart

sgomes, 2017-09-13:
After installing a UIPlugin from the Plugin Manager, the user is given the option to "Reload Now". However, if the user chooses to do so and the plugin renders its own templates into the current window (i.e., if the plugin's `will_handle_ui` returns True), a TemplateNotFound error takes place:

```
TemplateNotFound: dummy_mobile_ui_index.jinja2
```

If the server is fully restarted, though, no such error is produced and the plugin is able to find the template and render it normally.

Perhaps the "Reload Now" option should not be shown, and instead the user should be prompted to fully restart OctoPrint?

ntoff, 2017-09-13:
If you're the author, add this http://docs.octoprint.org/en/1.3.4/plugins/mixins.html#restartneedingplugin to tell the user a full restart is needed.

sgomes, 2017-09-13:
Ah, interesting, I had missed that mixin! Yes, I think it makes sense to push the responsibility onto the plugin author in this case, as they may not be using the built-in templating system at all.

I'll close the issue, thanks!

closed 2017-09-13 by sgomes