OctoPrint/OctoPrint #2115: [REQUEST] Please bundle all assets required

digilord, 2017-09-15:
#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.]

1. We lost Internet access. I thought I would get some models printed. OctoPrint requires two packages from CDN. Without these assets OctoPrint will not function.
2. mix panel-2-latest.min.js is not stored locally but instead at can.mxpnl.com
3. raven.min.js is not store locally but instead stored at cdn.ravenjs.com

[If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

I expected to be able to use my device as a stand alone printer that didn't require internet access to function.

#### What happened instead?

The interface did not completely load and was unusable.

#### Did the same happen when running OctoPrint in safe mode?

[Try to reproduce your problem in safe mode. You can find information
on how to enable safe mode in the Contribution Guidelines.]

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (master branch)

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

OctoPi

[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]

#### Printer model & used firmware incl. version

Monoprice Select IIIP V2

[If applicable, always include if unsure.]

#### Browser and Version of Browser, Operating System running Browser

[If applicable, always include if unsure.]

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ. - I HAVE READ THE F A Q

ntoff, 2017-09-16:
Please fill out the *ENTIRE* template, including trying it in safe mode, and include ALL the requested log files.

I believe those javascript files are part of the usage analytics and tracking used by the full featured slicer plugin and are NOT a part of octoprint at all. https://github.com/kennethjiang/OctoPrint-Slicer/issues/107

foosel, 2017-09-28:
OctoPrint bundles all assets that are required and runs perfectly fine without a connection to the internet. Some plugins however might not - like the aforementioned Slicer plugin. I strongly suggest complaining to the plugin author instead, they are the only one who can fix that.

closed 2017-09-28 by foosel