OctoPrint/OctoPrint #2116: Control Octoprint with the marlin printer interface.

atc1441, 2017-09-16:
Hi. 
I just tougth about the possibilitys to control octoprint via the marlin printer interface.
Lets say you dont have the computer near by and you want to print the excact file again. Then you could go into the printer menu and select a new option like replay print or you could shutdown octoprint for example

I know how to add a menu in marlin and with that i can send commands to the uart port onto octoprint.
How is it possible to take these commands in octoprint and do what they should do. 
Just like system commands editor but with uart income control.

Have a nice day.

ntoff, 2017-09-17:
Since this isn't a bug report, you might want to add [request] to your title to keep the bot happy, it seems like you want to add a feature to octoprint to respond to commands your firmware outputs? There is an action plugin to do that: http://plugins.octoprint.org/plugins/actiontrigger/ though I've not used it myself so I'm unsure of what it's capable of, also I don't know how you'd handle user permissions (i.e. you need to be logged in to print if you have access control enabled, you can't log in from an LCD panel on the printer).

I feel like if you're capable of hacking Marlin and adding features to it, you could probably make it output "action:print" and then have a plugin see that and print the currently selected file.

atc1441, 2017-09-17:
Thank you for your answer / request tip.

the plugin is exactly what i was searching for, i just looked into the code and it is easy edible so i will try to make the plugin work with my idea :)

i don't have access control enabled so i don't tough about that mhh.

cp2004, 2021-07-16:
Action commands are a well-used thing, OctoPrint supports several of them. Plugins that extend these are usable, so this can be considered resolved.

closed 2021-07-16 by cp2004