OctoPrint/OctoPrint #2117: Restart commends wont work

mjc619, 2017-09-17:
my restart system and restart octoprint commands aren't working after updating/ switching printers

I am running this off my windows netbook, but the newer builds aren't allowing me to use the restar commands in browser, I rebuilt and updated my virtual environment and rebuilt a new clone and it still didn't work.

I'm putting in my log, settings are the same from when I started trying to fix it, but tools have been updated, (there was initial pip issues but was a separate issue, pip and plugins install/ uninstall fine, until a restart is needed)

[octoprint.log.txt](https://github.com/foosel/OctoPrint/files/1308786/octoprint.log.txt)

we warned, my restart commands are curse words, also new reloads are me trying to find out if shutdown command worked, also printer/camera was disconnected

please help or help me find an alternative method, I just need to restart octoprint/the computer, I have a startup script that does that, also have not ran as admin if that is an issue.

ntoff, 2017-09-17:
So what is your actual shutdown command that you're trying to send? Windows doesn't use the same command as linux, all I see in there are your swear words, no actual commands. What were they before?

>microsoft windows 64bit, python recommended 32 bit

How do you run octoprint?...

>I have a startup script that does that

... what are the contents of that script? Is it a batch file in your "startup" folder? A service wrapper? A scheduled task?

Lastly, how old was the old version of octoprint? Maybe it's worth just wiping it out and starting again?

mjc619, 2017-09-17:
> actual commands.
I thought they where terminal commands that shortcut to restarting octoprint, I think I know what my problem is now

> contents of that script?

batch file that just opens the octoprint.exe and a shortcut the camera software with a few arguments

last time I used this was in class in may, new printer but same netbook.

a fresh install on a pi or a wiped hard drive would probably be better, but need to find a restart code

ntoff, 2017-09-19:
>but need to find a restart code

If all you're after is help figuring out how to make octoprint restart, please use the mailing list or the google+ community links in the post by the git issues bot. 

If you really are posting a bug, in that you had some restart commands that worked prior to updating octoprint, edit your original post, remove the swear words from your log and put your actual commands back so people can test using the same commands you are.

Please provide the *actual* contents of your batch files, not just "I think they were some terminal commands".

If you're starting OctoPrint  manually, it's not going to be able to restart itself. The way it restarts under raspbian / other linux based operating systems is via the operating system's service manager, not by any sort of internal octoprint command.

closed 2017-10-11 by foosel