OctoPrint/OctoPrint #2118: After long (~2 day) print, web UI shows "server disconnected" on Pi Touchscreen. OctoPrint seen as offline from other device browsers.

devsfan1830, 2017-09-17:
[Original_Topx9.zip](https://github.com/foosel/OctoPrint/files/1308804/Original_Topx9.zip)
READ THE FOLLOWING FIRST:

If not already done, please read the "guidelines for contributing"
aka the Contribution Guidelines that are linked ^-- just up there
in the big yellow box.

Also read the FAQ: https://github.com/foosel/OctoPrint/wiki/FAQ.

This is a bug and feature tracker, please only use it to report bugs
or request features within OctoPrint (not OctoPi, not any OctoPrint
plugins and not unofficial OctoPrint versions).

Do not seek support here ("I need help with ...", "I have a
question ..."), that belongs on the mailing list or the G+ community
(both linked in the "guidelines for contributing" linked above, read
them!), NOT here.

Mark requests with a "[Request]" prefix in the title please. For bug
reports fully fill out the bug reporting template (if you don't know
where to find some information - it's all described in the Contribution
Guidelines linked up there in the big yellow box).

When reporting a bug do NOT delete ANY lines from the template but
those enclosed in [ and ] - and those please DO delete, they are
only provided for your information and removing them makes your
ticket more readable :)

Make sure any bug you want to report is still present with the CURRENT
OctoPrint version and that it does not vanish when you start OctoPrint
in safe mode - how to do that is also explained in the Contribution
Guidelines linked up there in the big yellow box.

Thank you!

(Before submitting your ticket, please delete this text up to and
including the line too - it's only here for you, we already know it
by heart ;))

----

#### What were you doing?

Using OctoPrint with Touch UI and Raspberry Pi 7" Touchscreen. Pi boots to a browser window and autologs me into Octoprint+TouchUI.

After a roughly 48 hour print, during which the Touchscreen remains black due to its auto sleep (?), I wake the touchscreen and am greeted with a "server offline" pop up. It cannot auto reconnect. Pressing the reconnect button does nothing. The window scrolls so the browser UI and touch screen is responding. I just cannot clear the error and regain control of the printer via OctoPrint. Web access from my desktop and mobile phone browser also fail. The only fix is to SSH into my Pi and issue a "sudo reboot". Works fine again after that. I'm using the latest Raspbian and OctPrint. The print itself still completes. May be an issue with the web interface?

Ideally provide exact steps to follow in order to reproduce your problem:

1. Upload and Print GCode (Attached, zipped)
2. Do not interact with web UI at all for the 2 day duration
3. Try to interact with Web UI on Touchscreen, try to load UI on other device browsers.

If you encountered a problem with specific files of any sorts, make sure
to also include a link to a file with which to reproduce the problem.]

#### What did you expect to happen?

OctoPrint web interface allow me to start another print

#### What happened instead?

OctoPrint web interface unresponsive with a "server offline" error. 

#### Did the same happen when running OctoPrint in safe mode?

Not tested.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint

OctoPi 0.14

#### Printer model & used firmware incl. version

Prusa i3 MK2S, Firmware 3.0.12

#### Browser and Version of Browser, Operating System running Browser

Chromium (I Think) for the Pi Touchscreen, OctoPi 0.14
Safari on iOS10
Chrome 61.0.3163.91 on Windows 10

#### Link to octoprint.log

https://pastebin.com/ZZHpZxvY

#### Link to contents of terminal tab or serial.log

None.

#### Link to contents of Javascript console in the browser

https://pastebin.com/mwYJmY4E

#### Screenshot(s)/video(s) showing the problem:

Since it is a long print I am often not present when error occurs, nor can I witness it occur.

I have read the FAQ.
Yes

ntoff, 2017-09-17:
You have a bunch of internal IP connections (192.168....) but then you have a couple of external IP connections at the bottom

>2017-09-10 19:39:52,532 - octoprint.server.util.sockjs - INFO - New connection from client: 
174.204.10.78

and

>2017-09-10 21:22:01,826 - octoprint.server.util.sockjs - INFO - New connection from client: 98.163.102.175

Are those connections from you accessing it externally? If not, make sure you secure the pi from external access if you don't need it, could be that someone accessed it and did something to it while you weren't looking.

devsfan1830, 2017-09-17:
Yes that is likely me accessing it remotely. I check in on it during these long sessions from work. They appear to be verizon and Cox ip's which are my phone and isp, respectively. I have haproxy configured to require a password which is unique and strong with random characters. The pi ssh password was also changed.

devsfan1830, 2017-09-17:
I setup haproxy i followed this : https://github.com/PrusaMK2Users/MK2_Tips_and_Tricks/wiki/OctoPrint-HAProxy-Config

foosel, 2017-10-02:
Can you still ping the Pi when this happens? If not it's not OctoPrint but rather the Pi or the underlying operating system (OctoPi?) at fault. OctoPrint can't be accessed unless there's a working network connection in place and it can't take care of that, that's the job of the OS.

devsfan1830, 2017-10-02:
Each time its happened i've still been able to SSH into it so I can issue the reboot command. I can still manipulate the touch screen GUI at the time. It just doesnt do anything when i click reconnect. Its not happened for a while but I've also taken a slight break from multi day prints.  Also each time the prints do finish so it seems like its just the web GUI that stops working if thats even possible. The underlying system seems to stay fully functional. So its not a total showstopper. Just a minor annoyance to need to reboot in those cases. Next time it happens I'll leave it alone and post whatever you need (additional logs, video, etc). Just let me know what is needed.

foosel, 2017-10-09:
These things would be interesting:

  * You said it doesn't connect via a browser from your desktop or mobile either - what happens instead? What error do you get? Does it just not connect at all or do you get an error message telling you that "the OctoPrint server is currently not running"?
  * Check via ssh if the haproxy and octoprint processes are running: `ps -ef | grep octoprint` and `ps -ef | grep haproxy` - note that both should return at least **2** results, since the `grep` command will always be returned. Report back the output of this.
  * While usually there is not much written to it, `/var/log/haproxy.log` might contain some hint after all. Note that there also might be an older version at `/var/log/haproxy.log.1` since the logs roll over. Both could contain the crucial hint.
  * Instead of `sudo reboot`, try these steps to recover:
    * First try `sudo service haproxy restart` - check if that fixes it, report back
    * If that doesn't help, try `sudo service octoprint restart` - check if that fixes it, report back
    * If neither of those helps, you may reboot.

kaeus, 2017-10-11:
I've been having a similar issue recently and have not yet figured out the issue. I don't use HA Proxy on the Pi itself, I have a separate proxy server that hosts a bunch of access to other stuff behind SSH. My case, I've been doing repeated 18 hour prints, and pretty reliably right once the print finished the OctoPrint Server has the same issue. I usually just reboot the Pi (power off/on) and it comes back up fine.

Any particular place I can search for logs to help provide more detail on this issue? It isn't breaking completely as I need to clear the print bed anyway between prints, but is a bit of an annoyance to have to power cycle every time.

foosel, 2017-10-12:
@kaeus when that happens, can you still connect via SSH/ping the machine? Or is it gone completely? I've read that some people are still experiencing issues with power saving on the network interface, hence my question.

foosel, 2017-10-26:
Closing due to lack of response.

closed 2017-10-26 by foosel

devsfan1830, 2018-05-16:
So. Its happened again. Hopefully this time we can nail down whats responsible. It quit during a much shorter print this time. Causing the printer to stay fully heated and stopped in the middle of the print. Cant say for how long exactly. Octoprint Anywhere says the printer was last seen 8 hours ago. Don't think that's accurate. But this time I hit reset on the printer to regain control and kill the heat. Now I can hopefully provide useful info and logs.

Again, I can reach the Pi via SSH. But web ui via any computer and the pi touchscreen wont respond. Says the server is offline. , I use Octoprint Anywhere for remote access and it will not connect. Printoid I use for my phone access while home since the printer is now in my basement. That too will not connect. 

So, via SSH i get the following:
"ps -ef | grep -i octoprint" outputs this:
pi       15441 14056  0 21:54 pts/0    00:00:00 grep --color=auto -i octoprint

"ps -ef | grep haproxy" outputs the following:
root       925     1  0 12:10 ?        00:00:00 /usr/sbin/haproxy-systemd-wrapper -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid
haproxy    927   925  0 12:10 ?        00:00:00 /usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -Ds
haproxy    928   927  0 12:10 ?        00:00:06 /usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -p /run/haproxy.pid -Ds
pi       19468 14056  0 22:05 pts/0    00:00:00 grep --color=auto haproxy

"ps -ef | grep -i octoprint | grep -i python" returns nothing

-Worth noting I do not have haproxy configured anymore. This is a few months old reflash of octopi and i since closed off public internet access to my network, Hence Octoprint Anywhere just for basic remote view and i can kill a failed job.

Attached is the latest octoprint.log
[octoprint.log](https://github.com/foosel/OctoPrint/files/2011010/octoprint.log)

Using "sudo service octoprint restart" did indeed restart octoprint without a full reboot.

foosel, 2018-05-17:
@devsfan1830 I'm leaving this one closed and will rather look into this in your newly created ticket. Based on the information that was provided here it's impossible to say if you are even seeing the same symptoms as reported in this ticket or not since it was completely unclear whether it was just a network connection issue or a full blown crash (in your case it definitely appears to be the latter).