OctoPrint/OctoPrint #2119: "Download & Setup OctoPrint" links to dd. ther is somthing wrong with dd on linux...

pavia, 2017-09-18:
1. Download & Setup OctoPrint
2. Trying to erase sd with dd on ubuntu. Had to by a new sdcard
3. i fixed the problem using  **Etcher-1.0.0-linux-x64**
Pleas put a link to it in Download & Setup OctoPrint

I have read the FAQ 
sorry

Kunsi, 2017-09-18:
OctoPrint itself doesn't refer to using dd, installing it is just some play around using git and python.

Are you referring to [OctoPi](https://github.com/guysoft/OctoPi)? If yes, please report the issue there. If not, please report where that documentation error is located, what error message you got and how exactly you fixed it.

ntoff, 2017-09-19:
>Trying to erase sd with dd on ubuntu. Had to by a new sdcard

You don't use dd to erase an sd card, if there is an issue with your version of dd bricking your SD c ard, then you should post your issue in the bug tracker of whichever linux distro you're using.

As for the link to etcher, the official install instructions mention etcher http://octoprint.org/download/

>Unzip the image and install the contained .img file to an SD card like any other Raspberry Pi image.

If you follow the link, it takes you to the raspberry pi foundation's website and instructions on how to write an SD image *including using etcher*.

foosel, 2017-09-28:
Even if OctoPrint's documentation was referring to `dd` somewhere (which it doesn't), it would be correct - `dd` **is** the de facto standard tool for flashing cards under Linux/Unix, works fine if used correctly and doesn't require a 100MB download like Etcher does.

However, OctoPrint is not referring to `dd` in the official docs, it links to the official Raspberry Pi instructions instead (which as already pointed out refer the user to Etcher and only mention `dd` in "advanced instructions"), which work fine.

closed 2017-09-28 by foosel