OctoPrint/OctoPrint #2120: EventHandlerPlugin only returns filename for path and name on FileSelected

platsch, 2017-09-19:
#### What were you doing?

[Please be as specific as possible here. The maintainers will need to
reproduce your issue in order to fix it and that is not possible if they
don't know what you did to get it to happen in the first place.

Ideally provide exact steps to follow in order to reproduce your problem:

1. Create test plugin with EventHandlerPlugin
2. Select file
3. Access the `path` field in the payload dict

#### What did you expect to happen?
The [documentation](http://docs.octoprint.org/en/master/events/index.html#sec-events) states the `path` field should contain the path to the file which was just opened

#### What happened instead?
Both fields (`name` and `path`) return only the filename

This might be related to https://github.com/foosel/OctoPrint/issues/1963 but I think this is a different problem. I didn't carefully dig through the code yet to track down the issue, the dict is generated in standard.py (https://github.com/foosel/OctoPrint/blob/devel/src/octoprint/printer/standard.py#L1207)

#### Did the same happen when running OctoPrint in safe mode?
Not applicable, plugins are disabled in safe mode

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.4.0.dev544+gadbcc07 

#### Operating System running OctoPrint
Linux (Ubuntu, Debian) and Windows 10

#### Printer model & used firmware incl. version
Not applicable

#### Browser and Version of Browser, Operating System running Browser
Not applicable

#### Link to octoprint.log
https://gist.github.com/platsch/8c09437bae4698b9bc60b7a7dfde098e

#### Link to contents of terminal tab or serial.log
Not applicable

#### Link to contents of Javascript console in the browser
Not applicable

#### Screenshot(s)/video(s) showing the problem:
Not applicable

I have read the FAQ.

foosel, 2017-09-28:
The returned `path` is supposed to be relative to the file's storage location ("the file’s path in its origin storage location" - admittedly worded a bit suboptimal here). So if you are looking at a file stored at a location's root folder, both `path` and `name` will return the same:

```
/
|- test.gco <- name=test.gco, path=test.gco
|- subfolder/
   |- test2.gcode <- name=test2.gco, path=subfolder/test2.gco
```

Is this what you are seeing?

platsch, 2017-09-29:
Yes, that's exactly what happens and now I better understand the documentation, thx.
I ran into this when updating the deprecated `file` argument in a `FileSelected` events, which returns the full path on disk (and I admit it is well documented there).
So what am I supposed to use to actually open the file then, the `generic global_get_basefolder` method + path? Is there any reason I don't see not to provide a full path parameter for convenience?
I would update the documentation with a more extensive description once I fully understood the context :-)

foosel, 2017-10-02:
> Is there any reason I don't see not to provide a full path parameter for convenience?

Depending on the storage the file resides on, there might not be a full path that you even can access locally. E.g. in the case of SD files. The storage sub system is modular, and even though not even the SD card stuff makes use of that yet, the goal is to allow everything plus the kitchen sink to be registered here and somehow provided to the printer to print. So local paths are kinda optional.

> So what am I supposed to use to actually open the file then

If a plugin or internal component consuming such an event needs a path on disk for a file (and one is available), it can be retrieved via the file manager ([`path_on_disk(storage, path)`](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/filemanager/__init__.py#L597)). If no such local file is available, that should throw an `io.UnsupportedOperation` exception in the associated [`StorageInterface` instance](http://docs.octoprint.org/en/master/modules/filemanager.html#octoprint.filemanager.storage.StorageInterface.path_on_disk). I just noticed that the documentation for file manager itself is missing, I'll have to fix that. 

Caveat: as hinted at above the SD card stuff currently isn't implemented as a `StorageInterface` (keep meaning to do that and then other stuff gets in the way). Which means that if you do `path_on_disk("sdcard", some_path)` you'll get a `NoSuchStorage` exception back. Which in that case you can easily handle.

platsch, 2017-10-09:
Thank you for the extensive explanation, that solves it.

closed 2017-10-09 by platsch