OctoPrint/OctoPrint #2122: [Request] Allow multiple print instances simultaneously on one pi3 running octoprint out of the box..

dkingsjr, 2017-09-19:
Hello,
I have tried looking for the request, but cant find one, so I am starting a new request...

I would like consideration, to allow multiple print instances to take place using ONE raspberry pi. considering that there is 4 USB Ports on the pi, I figure, why not? Currently, it is only possible to run ONE instance. It would be nice to run 4 printers off of one pi, and without a noob like me having to edit the sourcecode and possibly messing something up. I myself have no intention to get into programming, but I and I'm sure many others, would love to do what I'm asking. It would be nice having this feature run out of the box. I myself envision the possibility to connect a screen to the pi via HDMI, bringing up 4 browser instances all with the octoprint server running, but with different printers cued in each, ready to print simultaneously. It would also be convenient, as we would have all our files on ONE server for 4 printers to use.

Thanks for the consideration!
Dave

ntoff, 2017-09-20:
The M33Fio plugin has an instance manager I think, it does add a lot of extra stuff as well that you may not necessarily need (you don't need an M3D printer for the plugin to work either) but it adds a little drop down box that lets you start a new session.

You might run into bandwidth issues though, I'm pretty sure those 4 usb plugs all share the same USB bus back to the main chip. Give it a try and see if the printer(s) start acting funny the more of them you have running.

https://github.com/donovan6000/M33-Fio

foosel, 2017-09-28:
It is already possible to run multiple instances, you just need to specify a different base folder for each of them via the `-b` or `--basedir` parameter, and a different port for each of them via the `--port` parameter, e.g. `octoprint serve -b ~/.octoprint2 --port 5001` for a second instance, `octoprint serve -b ~/.octoprint3 --port 5002` for a third and so on.

See also `octoprint --help` and `octoprint serve --help` and [this blog entry by Sebastian Andrivet](http://sebastien.andrivet.com/en/posts/installing-multiple-octoprints-raspberry-pi-3-running-raspbian/).

Sharing of files across instances is currently NOT supported due to how metadata is managed, however that is already tracked in #1488

closed 2017-09-28 by foosel