OctoPrint/OctoPrint #2124: Invalid time left / progress on relatively big file

zsoltm, 2017-09-21:
#### What were you doing?

Printing .gcode from local storage, uploaded via WEB interface (printer's SD support is disabled).
The .gcode I'm printing is relatively large, 64MB.

1. upload big 64MB .gcode file -- estimated to a bit more than 12hours by slicer
2. select it from "files" section
3. press print button
4. go to work, watch a movie (in short let almost 12 hours pass)
5. open Octoprint's main page

#### What did you expect to happen?

Progress indicator at "state" area shows accurate percentage: ~98% and relatively accurate time remaining (print time left): ~5minutes.

#### What happened instead?

Progress indicator and time left shows roughly half of the actual print state: progress has 49%; print time left has 12 hours. In reality print is almost finished, currently printing the last few layers.
Also "print time left" shows accurate reading: 11:52:27.

#### Did the same happen when running OctoPrint in safe mode?

Not checked.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (HEAD -> master branch)

#### Operating System running OctoPrint

Armbian 5.31 on an old BananaPI.

#### Link to logs / .gcode / screenshot

https://drive.google.com/drive/folders/0B5crNvyJYEJbMEtPWVNpMVZaTVE

#### Link to contents of Javascript console in the browser

https://pastebin.com/pqRBCMnC

foosel, 2017-09-21:
A quick look at your file reveals roughly 32MB of `\0` bytes at the end of it. It's this the exact file as generated by your slicer? If so OctoPrint's behaviour is correct here. Since at the end of the print it only has read half of the file (the other half is garbage), it has to assume it will still need about the same time again. It can't know that the rest of your file consists only of `\0` bytes. I recently saw another person on Twitter run into the same thing with S3D - might be a bug in that slicer. If you strip away the trailing 32MB of `\0` bytes, your file should behave as expected (and also take up less space obviously).

While it is possible for OctoPrint or rather plugins to preprocess files that are uploaded to eg strip comments or trailing whitespace and such, OctoPrint by default does NOT do that in order to preserve the files that are uploaded as is. But working around a slicer producing such trailing data would be possible via a plugin.

zsoltm, 2017-09-21:
@foosel You're right, indeed the .gcode is padded with zeroes. Wow, that's totally unexpected. Shouls've been suspicious that the file is exactly 64MB, but I thought it's a lucky coincidence. Yes it's exactly how it is generated by S3D. I still have the .factory file, it's re-producible. Whenever I save the .gcode from it it's padded to 64MB. I'll check what's going on, and maybe report it to them if nobody did it so far.

Totally agree OctoPrint is behaving correctly in this situation. Pre-processing the file "just in case it has garbage in the end" wouldn't make much sense, waste of processing time, size based progress calculation is accurate enough.

Feel free to close this, thanks for taking the time checking this out.

foosel, 2017-09-22:
Alright, closing this then :)

Would be great if you could follow up with your findings though in case others stumble over this. If it's a general S3D problem I'm sure that will happen. Btw, what version of S3D is that? I'm not seeing this on my own license but still haven't upgraded to the new version they released some weeks ago.

closed 2017-09-22 by foosel

zsoltm, 2017-09-22:
Yes it's the latest version, 4.0.0. I went trough a few rolls of filament with the earlier version, didn't have any issues either.

Kunsi, 2017-09-22:
Maybe provide the needed files to reproduce the error in here, so we may check if the error persists across computers (thus resulting in a simplify3d bug).

zsoltm, 2017-09-23:
@Kunsi sure, I uploaded the .factory file to the gdrive space I linked above, so you could try. To reproduce: open .factory, "preparte to print", "save toolpaths to disk". The resulting .gcode for me is always padded to 64M. It's a Win10pro system. I no longer have my Mac to try.

adammhaile, 2018-01-31:
FYI, looks like this is a bug in Simplify3D 4.0.0 that was fixed in the latest release: https://forum.simplify3d.com/viewtopic.php?f=9&t=8459
For whatever reason they are literally padding any gcode files over 32MB with zeros. Latest version shouldn't do that anymore.