OctoPrint/OctoPrint #2126: [Request] A way to exclude specific areas to recover from failure on multi part prints

smokingfish, 2017-09-25:
Hello there,

I just finished a nice multi part print using OctoPrint on my Pi3.
After about 3 hours I had one part that failed. I managed to recover the rest using a deck of cards and it continued printing, missing no time and just a little bit of material for the one failed part.

I guess there is no way to distinguish between models once the gcode is created, so cancelling just one would not work.

But maybe it's feasible to set an area to exclude while the print is still running.
That can be as simple as a command or even integrated into the gcode viewer. Once you see that one part fails, you select an rectangle/circle around in the viewer and exclude it. All gcode commands in that area would just be ignored. After that you can do a little cleanup and all the other parts will just continue printing as normal.

Is this possible?

Best regards,
Marcus

hazetekllc, 2017-10-25:
I have had this happen also I was printing a chain and just one failed it would of been nice to have this feature to stop that one and let the other 19 finish instead i had to baby sit it for 2 hrs. I Plus 1 this request

crafterix, 2018-04-05:
This would be nice! +1

smokingfish, 2018-06-21:
For anyone searching: there is now a nice little plugin by @paukstelis:
[link](https://github.com/paukstelis/OctoPrint-Cancelobject)

I would still love to see it integrated into the gcode viewer and without the markers, but this a great step in the right direction!