OctoPrint/OctoPrint #2127: Start at center not working

johnryanlee, 2017-09-26:
#### What were you doing?

Trying to print something

#### What did you expect to happen?
The print would start in the center of the bed

#### What happened instead?
The print started at the front-left

#### Did the same happen when running OctoPrint in safe mode?

I am not aware of safemode and therefore did not try

#### Branch & Commit or Version of OctoPrint

0.14

#### Operating System running OctoPrint

OctoPi 0.14

#### Printer model & used firmware incl. version

Geeetech Prusa I3 W (GT2560 board)
Marlin 1.1.5

#### Browser and Version of Browser, Operating System running Browser

NA

#### Link to octoprint.log

Don't think this is necessary

#### Link to contents of terminal tab or serial.log

Don't think this is necessary

#### Link to contents of Javascript console in the browser

Don't think this is necessary

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

Don't think this is necessary

I have read the FAQ.

ntoff, 2017-09-26:
>Don't think this is necessary

The logs *are* necessary to see what OctoPrint is sending to your printer. Without those, we can't help you. Please run another print (or run a print in dry run mode if your firmware supports it and you don't wish to waste filament), and provide *all* the requested information. Also please enable the serial log and post that, then we can see the co-ordinates that OctoPrint is sending to your printer.

foosel, 2017-09-26:
See also my comment here https://github.com/foosel/OctoPrint/issues/2128#issuecomment-332218223

foosel, 2017-09-28:
@johnryanlee can this be closed as well? As mentioned in #2128 the positioning on the bed is the job of your slicer, and there's not much OctoPrint can do here if that's not working properly.

johnryanlee, 2017-09-28:
yes thanks

closed 2017-09-28 by johnryanlee