OctoPrint/OctoPrint #2128: [request] Offset parameters

johnryanlee, 2017-09-26:
#### What were you doing?

Printing

#### What did you expect to happen?

Print to occur on bed

#### What happened instead?

Print runs off bed

#### Did the same happen when running OctoPrint in safe mode?

Don't think this is relevant

#### Branch & Commit or Version of OctoPrint

OctoPi 0.14 (downloaded on Saturday)

#### Operating System running OctoPrint

OctoPi 0.14 (downloaded on Saturday)

#### Printer model & used firmware incl. version

Geeetech Prusa I3 W DIY (GT2560)
Marlin 1.1.5

#### Browser and Version of Browser, Operating System running Browser

NA

#### Link to octoprint.log

Don't think this is necessary

#### Link to contents of terminal tab or serial.log

Don't think this is necessary

#### Link to contents of Javascript console in the browser

Don't think this is necessary

#### Screenshot(s)/video(s) showing the problem:

I think it's obvious enough - imagine, printer prints on an air bed, it ain't gonna work... nuf said

ntoff, 2017-09-26:
First of all, the logs are requested for a reason (please include them), and also please enable the serial log and post that so we can see exactly what is being sent to your printer.

>I think it's obvious enough - imagine, printer prints on an air bed, it ain't gonna work... nuf said

What are you even talking about? OctoPrint sends gcode to your printer, if your slicer isn't producing the correct gcode then that's not OctoPrint's fault. If your printer's firmware isn't set up correctly, OctoPrint isn't to blame for that either.

Please don't be so arrogant in your posts, if you think there really is a bug, please include all the information that is asked of you (like I said, it's requested for good reason whether you think it is or not).

johnryanlee, 2017-09-26:
@ntoff Apologies for my comment.  I was not trying to be arrogant, it was meant to be humorous sarcasm.  The software is awesome and I am grateful to be able to use it!  I will treat this professionally going forward.

I will do as you say tonight when I can test again.

foosel, 2017-09-26:
Just for the record, positioning of your printed objects on the bed are between your slicer and your printer. OctoPrint only pumps GCODE produced by your slicer to your printer, it doesn't modify or add anything. See also https://github.com/foosel/OctoPrint/wiki/FAQ#octoprint-is-telling-my-printer-to-do-wrong-things. This also applies to your other ticket #2127

johnryanlee, 2017-09-26:
Does the export of the Cura profile include printer information?  I see there are start/end GCODES in the profile file which I removed in an attempt to get this issue sorted out.

Please excuse me, I've only been at this for 2 days and only having a printer for 5 - made sure it worked and now started hacking it.

My reasoning for this being a "request" is that I feel the firmware just handles the "how to position" and failsafes while the software handles the "do this" and "do that".  So a start position or an offset/override could be configured through OctoPrint.  In my research I do see that there are codes to override stuff so that means OctoPrint can do it and does provide the feature through job statuses (start/cancel/end/complete; etc), but I don't know how to use that yet.  I am studying the RepRap wiki but perhaps you could assist.

johnryanlee, 2017-09-26:
Recv: echo:busy: processing
Recv:  T:190.26 /190.00 B:69.99 /70.00 @:27 B@:127 W:0
Recv: ok
Send: N8 M107*45
Recv: ok
Send: N9 G0 F4800 X-9.897 Y5.298 Z0.200*57
Recv: ok
Send: N10 G1 F1800 X-18.900 Y5.298 E0.22458*46
Recv: ok
Send: N11 G1 X-18.900 Y-35.784 E1.24938*90
Recv: ok
Send: N12 G1 X-18.852 Y-36.778 E1.27420*88
Recv: ok
Send: N13 G1 X-18.685 Y-37.907 E1.30267*89
Recv: ok
Send: N14 G1 X-18.406 Y-39.028 E1.33149*81
Recv: ok
Send: N15 G1 X-18.025 Y-40.096 E1.35977*93
Recv: ok
Send: N16 G1 X-17.537 Y-41.131 E1.38832*87
Recv: ok
Send: N17 G1 X-16.948 Y-42.115 E1.41692*92
Recv: ok
Send: N18 G1 X-16.268 Y-43.032 E1.44540*86
Recv: ok
Send: N19 G1 X-15.503 Y-43.878 E1.47385*84
Recv: ok
Send: N20 G1 X-14.658 Y-44.648 E1.50237*86
Recv: ok
Send: N21 G1 X-13.737 Y-45.334 E1.53102*81
Recv: ok
Send: N22 G1 X-12.756 Y-45.920 E1.55952*80
Recv: ok
Send: N23 G1 X-11.712 Y-46.418 E1.58838*87
Recv: ok
Send: N24 G1 X-10.642 Y-46.799 E1.61671*86
Recv: ok
Send: N25 G1 X-9.540 Y-47.077 E1.64506*110
Recv: ok
Send: N26 G1 X-8.395 Y-47.249 E1.67394*99
Recv: ok
Send: N27 G1 X-7.389 Y-47.300 E1.69907*98
Recv: ok
Send: N28 G1 X23.885 Y-47.300 E2.47920*123
Recv: ok
Send: N29 G1 X24.879 Y-47.250 E2.50403*112
Recv: ok
Send: N30 G1 X26.008 Y-47.084 E2.53249*116
Recv: ok
Send: N31 G1 X27.115 Y-46.807 E2.56096*126
Recv: ok
Send: N32 G1 X28.192 Y-46.425 E2.58947*122
Recv: ok
Send: N33 G1 X29.228 Y-45.936 E2.61804*120
Recv: ok
Send: N34 G1 X30.210 Y-45.351 E2.64656*123
Recv: ok
Send: N35 G1 X31.133 Y-44.668 E2.67520*118
Recv: ok
Send: N36 G1 X31.981 Y-43.899 E2.70376*112
Recv: ok
Send: N37 G1 X32.749 Y-43.054 E2.73224*116
Recv: ok
Send: N38 G1 X33.430 Y-42.137 E2.76073*119
Recv: ok
Send: N39 G1 X34.019 Y-41.157 E2.78925*127
Recv: ok
Send: N40 G1 X34.511 Y-40.124 E2.81780*126
Recv: ok
Send: N41 G1 X34.898 Y-39.044 E2.84641*115
[... 193 lines cut ...]
Send: M140 S0
Recv: ok
Send: M106 S0
Recv: ok
Send: M117 Job cancelled!
Recv: ok
Send: M105
Recv: ok T:190.00 /0.00 B:68.79 /0.00 @:0 B@:127
Send: M105
Recv: ok T:189.66 /0.00 B:68.60 /0.00 @:0 B@:0
Recv: echo:enqueueing "G28"
Recv: echo:busy: processing
Send: M105
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: echo:busy: processing
Recv: X:0.00 Y:0.00 Z:0.00 E:15.75 Count X:0 Y:0 Z:0
Recv: ok T:185.42 /0.00 B:65.30 /0.00 @:0 B@:0
Send: M105
Recv: ok T:185.42 /0.00 B:65.30 /0.00 @:0 B@:0
Send: M105
Recv: ok T:183.85 /0.00 B:64.81 /0.00 @:0 B@:0
Send: M105
Recv: ok T:180.82 /0.00 B:64.17 /0.00 @:0 B@:0
Send: M105

johnryanlee, 2017-09-26:
I managed to figure out the X axis offset and saved to EEPROM
M206 X-20, what I actually needed was M206 X-40
Then M500 to save
Then M501 to recall to confirm

Now what i'm finding is that when it's filling in, it's moving straight instead of diagonal and the X axis is showing what the Y axis is doing.  However on the control screen, everything works as it should.

This issue can be closed.  As much as it would be a nice to have it's apparent that the setting should be in the on the printer.  Apologies for coming across as arrogant earlier.  It was never my intention.

johnryanlee, 2017-09-26:
Issue is the slicer. most likely the slicer profile.
I saved the slice from the easyprint 3d software. Uploaded to Octopi and it worked. 

A lot of knowledge gained today and hopefully not enemies.

ntoff, 2017-09-27:
>This issue can be closed.

You have the power to close your own issues, if you really are done with it then you should be able to close it yourself (to save foosel the trouble).

Also if you require assistance, you may wish to use the G+ group or the mailing list and save this place for bug reports as your issue seems to be more one of how to set up your slicer to match your printer than some sort of bug in the way OctoPrint works (given your other thread).

https://groups.google.com/group/octoprint

https://plus.google.com/communities/102771308349328485741

Or there's always reddit https://www.reddit.com/r/3Dprinting although that's more of a general 3d printing forum, it's not dedicated to octoprint, but it might yield some info if you need help setting up your slicer.

foosel, 2017-09-28:
Closing this as requested. No enemies gained I think ;)

closed 2017-09-28 by foosel