OctoPrint/OctoPrint #2130: [Brainstorming] Use a machine learning model for print time estimation.

radicalbiscuit, 2017-09-29:
Machine learning is tomorrow's promise of less explicit coding for features that are more effective and accurate than conventional means. One of the difficulties in developing a machine learning model, however, is the gathering of massive amounts of data.

I believe the OctoPrint community would be an excellent source of machine learning data. A plugin or feature could allow users to opt-in to (or -out of) a data gathering project whose corpus would be used to develop a ML model, the focus being to develop an accurate print time estimator. Even with just the gcode and actual print time, a decent model could be trained, not to mention the performance that might be achieved with further attributes (e.g. extruder type, filament type, etc.).

ML requires a lot of processing power in conjunction with data to train the model, but once the model has been developed, it's easily distributed and, for something like this, pretty efficient. I think it's an excellent ML application.

I'm looking for input regarding feasibility, likelihood of obtaining opters-in (or piggybacking on something like the OctoPi project), most appropriate learning models, whether this has been explored before (I found nothing in my admittedly cursory search), etc.

TTN-, 2017-10-21:
This is a very interesting concept, my biggest question around this is will we be able to see this without too much of a performance hit? 

In principle, the ETA is simply comprised of a few components:

- Heating time of the printer (just grab the average of the last few heat times)
- Print/travel moves (we must factor in acceleration, jerk, and maximum movement speeds of each axis. This includes retraction moves)

ML is totally possible but it does seem a little over engineered. 

If it works, its not stupid. If it can be done efficiently, and someone is happy to engineer it, why not?

foosel, 2017-10-23:
Sorry, I somehow missed this...

I understand the proposal such that the ML itself would be done on a central server and OctoPrint would send data there for analysis, and also for requesting estimates for a provided file (preferably offline by acquiring the trained model?).

First of all, I'm all for experimenting, and if someone with a bit of a background in ML wants to give this a shot I'll certainly encourage this and try to lend a hand where possible. 

Personally I think though that the setup and maintenance of the necessary infrastructure would be a bit too much of an overhead and commitment compared to what would be achieved. Something like a static correction factor derived from past prints on the same printer across all models would probably achieve something very close to that (or having users enter acceleration/jerk settings or querying that from the firmware).

radicalbiscuit, 2017-10-23:
I'm a big proponent of recording all ideas. Ideas are fluid, and a bad idea can lead to a better one. This is probably a bad idea, as I don't think a machine or human could look at gcode and "intuitively" know how long it would take to print. An iterative list of instructions is far more susceptible to the method already employed. It's not well suited for machine learning.

This might be a better idea for slicers, estimating print time before generating gcode by looking at the model and taking a few parameters such as planned layer height, etc. Regardless, I don't think this is the outlet for my constant search for a novel use of machine learning through which I can learn machine learning.

closed 2017-10-23 by radicalbiscuit

eubr, 2017-12-16:
**TLDR**: This is very easy, as long as we can ask users for a bunch of print times and gcode pairs to train the model. Printing time with and without heating would be the best, but if not, that's probably ok too, as long as prints aren't way too short. We don't need very many for this simplistic model to work.

Long version with detailed explaination:
I don't think any kind of infrastructure would be necessary for a simple model. We're talking about a couple dozenfloating point operations... let me explain.

Basically, the way this would work is this:
1. the training of the model that will estimate times is done on someone's workstation. The training consists of:
1.1 printing many gcode files and recording how much time they each took to print
1.2 choosing some things (we call them features) we think might be somehow indicative of the actual time whatever the gcode file we're printing is: number of lines, number of moves, size of the file, estimated layer height, how many times a move significantly changes the direction of the tool times the number of moves... this kind of thing. It **does not matter** that we know how these things influence the printing time. The whole job of the training process is to let the model figure this out for itself.
1.3 now we run the training on the workstation. Basically, the training algorithm will take a look at all those features we chose, and try to understand how the final printing time we recorded is influenced by the features we chose. For the particular problem of estimating printing time better than manual heuristics, we can probably train an off the shelf model from scikit learn (which is a python library!), and it will have a pretty decent accuracy. When the process is done (if we chose a linear model, this will take seconds), we have a model we can embed in octoprint. The actual model is going to look something like this (we can obviously put the data in a separate data or config file):

```
def get_printing_time_in_seconds(features):
  return (features.number_of_moves * 0.176 +
          features.number_of_direction_switching_moves * 0.42 + 
          features.bias * 244.23
  )
```

as you can see, this is just a couple dozen floating point ops. nothing to worry about!

2. now we import something into octoprint and we want to know how much time it takes! we just compute the features, like the number of moves, pass them to the prediction function (like the example above) and we have an estimate in seconds.

If we don't trust this very much, we can just present our estimate as get_traditional_estimation() * 0.75 + get_ml_estimation() * 0.25. We are weighting this new prediction just a little bit, and if the feedback from the users is positive, that is, the predictions are getting better, we can crank it up to 50%, maybe eventually to 100%.

I believe this would be fairly simple to implement. The only part that would take (for example) me any time is doing the actual printing. If we can get people to give us their gcode and print times, this would be basically a weekend project! It might or might not work but I believe it might be worth trying.

Erutan409, 2017-12-17:
Only read the TLDR; - but if the part about getting paired g-code means sending said data back to a centralized server, that might right into some potential legal issues.

My only reference point for this is Simplify3D. They really don't like it when people share their sliced g-code files.

radicalbiscuit, 2017-12-17:
Features could be extracted at the endpoints rather than sending the entirety of the gcode. The disadvantage is that the features would need to be decided upon entirely in advance, but it would avoid any privacy or legal issues involved with sending the unabridged gcode.

There could be other applications of machine learning to 3D printing and the "hard" part is beginning the data collection. OctoPrint has a large userbase and I think it's the best candidate for implementing an opt-in or opt-out data collection program with a default stats collection server. After data are collected, then we could start thinking about applications for the data.

TTN-, 2017-12-17:
I think we're over complicating it.

Given X acceleration and Y minutes for warmup we can calculate exactly how long a print will take.

When the print completed sooner than we expected, we increase the acceleration but only by a by say 80% of what we expected to need to increase it by. This way it won't overshoot its estimate and converge quickly to exactly the correct estimate.

radicalbiscuit, 2017-12-17:
Yeah, that's why I originally closed the ticket, because the specific application I suggested is not a good fit for machine learning. I tend to be excitable and got ahead of myself. Print times are already straightforward to calculate with relative accuracy because gcode is procedural and written by machines, for machines. As far as print estimates go, there's not a lot of intuition that needs to go into it.

However the continuing discussion has me thinking about other possible stats/machine learning applications of features that can be extracted from individual gcodes/prints and I catch myself once again putting the cart before the horse. The hardest part of any machine learning application is compiling the corpus of data. Applications often don't manifest until one can study the collected data and begin to develop some insights.

One very useful application could be the prediction of printer part failures. This wouldn't need gcode-level granularity of data, just some simple info about the printer and tracking its usage and failures. Then OctoPrint could suggest printing/sourcing specific replacement parts as the probability of failure rises. This, however, would require users keeping a log of failures and replacements.

Another might be a print problem diagnostic. Something that takes features of your gcode, or the gcode itself, along with the problem you're experiencing (lifting corners, shifting layers, etc.) and info about the printer (already entered in OctoPrint), and return a more specific set of suggestions than general troubleshooting guides online can provide.

But until you collect data, and lots of it, there's not much point in speculating what could become of it.

eubr, 2017-12-17:
About data collection: I was talking about asking people to send their data if they wanted. Nothing implemented in software, not opt in, not opt out, just "please send us some data if you want".

fitzterra, 2018-02-17:
This to me originally seemed like an interesting use of ML, but I did not give it too much thought because the current system of time estimation was good enough for me. This has changed lately since I am printing larger object more often.

The flow for me now is:
1) Slice large object using slic3r - already having to wait there for computing large objects
2) Upload to Octoprint straight from slic3r - this is quick even for large prints and works very nicely, but... for large prints I can not tell slic3r to have the print start immediately on upload, because I need to wait for the print time estimator to examine the file and calculate the print time.
3) Wait for Octoprint to calculate the print time - this could take very long for complex objects. I've found that if I start the print before giving enough time for the print estimate to be calculated, it takes very long for this to be done while the print is running, and sometimes it does not even get calculated, or calculated very inaccurately.

So a question: 
From my understanding of some of the posts above, a sweep over the gcode file will be needed to calculate the "features" required by the model to make the estimation. Since this should be a straight forward enumeration of certain strings in the file, I assume that even for big gcode files this should be much faster than doing floating point calculations for each move - is this correct?
I know that the prediction from the model will be very fast, so is my assumption correct one advantage to this would be speed in calculating the print time?

eubr, 2018-02-17:
> I assume that even for big gcode files this should be much faster than doing floating point calculations for each move - is this correct?

Some floating point calculations for each move are going to be required anyway, because the model needs to know stuff like what the total travel distance is, how many moves abruptly change direction and so on.

I don't know how octoprint estimates the print time. If it does any fancy stuff, then it might be faster to do the sweep for the ML model. If it just does a couple of sums over the move times or something like that, then the ML model will not improve this.

Out of curiosity, when you say "very big gcode", how big are we talking about here? 10MB? 100MB? 1GB?

TTN-, 2018-02-17:
Generally very big for me would be on the order of a few hundred megabytes.
I've been printing for some time and biggest file i can remember would be
around 300MB of gcode.

arhi, 2018-02-17:
@TTN- can you run one of those 300M files trough https://git.bad-team.net/arhi/gcodestat I'd really appreciate that test :D ... how quickly will it go trough the file without --output and with --output, how accurate it will be (if you maybe know exactly how long it took for the file to print) etc.. mine gcode files are in the vicinity of 20-60mb so nothing even close to 300

@radicalbiscuit i don't really see machine learning as a solution here as to get accurate timing of a g-code file you just need to know few firmware parameters and you can process g-code rather fast (under second for my normal g-code file size).. I for e.g. made a g-code postprocessor that process the g-code file and then adds M117 every X% of the time so you get that info (remaining % and time) on your display in some predefined steps (by default 10% but you can use 1% if you want it's configurable).. I lack a way to push that info into octoprint (request here https://github.com/foosel/OctoPrint/issues/2359 ) so I just use a plugin to display M117 info on web page .. anyhow that processing could be added to octoprint itself should not be a biggy

fitzterra, 2018-02-20:
Thanks for the replies.

Summary:
Without having properly investigated this yet, it seems that for larger files (5GB+) the current time estimator takes quite long, and that it gets even worse to get accurate **Print Time Left** values when the print is started before the time estimation has completed.

I was hoping that using some ML model, and passing it the raw gcode, without first doing lots of floating point ops (which I am making an assumption is the reason the estimation takes so long), could speed up the flow from getting the gcode on the printer, and the print starting.

More (TL;DR):
@eubr 
> Out of curiosity, when you say "very big gcode", how big are we talking about? 10MB? 100MB? 1GB?

The files that  I have a problem with as far as time estimation goes are in the order of 5MB and larger. Not very large I suppose, but from my observations in estimating the **Approximate Total Print Time**, this may be a bit taxing on the RPi. From what I have noticed (seat of the pants, not scientific), once the print is started, it takes even longer to get this calculation done. I have not delved into how this is calculated or what happens when the print is started while calculating the estimated time, but I assume running the print gets preference in CPU cycles, which could have an impact on the estimation calculation.

@arhi : There is the "Detail Progress" plugin (https://github.com/dattas/OctoPrint-DetailedProgress) which does M117 messages to report progress on the fly - maybe something you have not seen yet?

Anyway, this plugin seems to use an internal `_printer.get_current_data()` method to update the progress, which I suspect is the same method used for the **Print Time Left:** field on the web UI. For a print that has started before the estimated print time has been calculated, it _seems_ that this "current data" can be very inaccurate - just a feeling, and no proper testing has been done to verify this.

Conclusion: lots of assumptions and seat-of-the-pants measurements here, so best would be to do proper testing to understand the reasons I do not get accurate time remaining estimates for larger prints :-)

arhi, 2018-02-20:
@fitzterra that plugin sends what octoprint calculated as progress to the printer, but that calculation is flawed so not much useful imo... the error you get is 'cause acceleration, jerk, junction deviation and other firmware settings are not taken into consideration for time calculation, only the speed and length from the g-code is... that's why I made https://git.bad-team.net/arhi/gcodestat as it does exactly that, calculates time using firmware data too. I'm running smoothieware and for smoothieware (acceleration, junction deviation and max speed) I get time calculation discrepancy (calculated vs real) on 3 different printers below 0.1% ... have not tried marlin (I don't run marlin) but acceleration part is same, marlin uses jerk and not junction deviation so that part is not same so with marlin I expect bit more then 0.1% descrepancy, anyhow it will be few orders of magnitude more precise then what currently octoprint assumes or what slicer's like slic3r or simplify3d estimate :)

now since my knowledge of python is very rudimentary (I rather dislike it) I have not tried to embed this into octoprint myself, but I run a gcodestat as postprocessor to g-code generation and I embed M117 into the file with "real time to finish" calculation... 

as for the time calc itself, it's rather simple, I doubt it would be a problem for octoprint to do it pre-printing... look at here: https://git.bad-team.net/arhi/gcodestat/blob/master/calcmove.c#L108
it's rather simple math (2x using sqrt() but these arm's should handle that without a problem) ..  for e.g. 200M gcode file takes about 63seconds on "orange pi one" .. not super fast but far from too slow (one can run it during preheat for e.g.)