OctoPrint/OctoPrint #2131: [Request] Octopi: Warning text on shutdown confirmation dialogue

scottpost4, 2017-09-29:
Simple request for the addition of warning text on the shutdown confirmation dialogue box. 
"Shutting down the system will stop/cancel any printing currently underway." 
I was using Octopi the other day simply to MONITOR my printer. The print was started from the printer, and was on an sd card mounted in the printer. When the print was almost complete I shut down the raspberry pi using the shutdown system command in the web interface. I received no warning (as I usually would with most other tasks that would interrupt prints). I assume Octopi, when shutdown, sends a disable steppers or stop command to the printer. This stopped my nearly done 6 hour print. This text would have given me pause, and I would have left the system running until the print was complete.

ntoff, 2017-10-02:
>I received no warning

Do you mean it didn't ask you at all if you wanted to shut down? Or it asked you if you wanted to shut down, but just didn't mention interrupting a print?

>I assume Octopi, when shutdown, sends a disable steppers or stop command to the printer. 

Not 100% sure but I'm pretty sure it doesn't. Some printer mainboards just like to reset themselves when they see certain types of serial traffic. The board probably just thought "hey, serial connection, reset time!". It's how arduino / some atmega based boards are so easily able to be flashed without requiring external programmers. Any serial connection will reset the board, putting it into a sort of "gimme the firmware" mode for a short while.

scottpost4, 2017-10-02:
So to answer the first question, no warning it would interrupt prints. And for the second question, you're right it could have been the printers controller stopping because of the traffic it received. 
The screenshot looks perfect! Exactly what I imagined. (:

galamdring, 2017-10-04:
The process of reconnecting to the serial bus of the controller resets the controller. This isn't something done by Octoprint, but the Arduino, as it takes every connection as a possible programming connection, and restarts the bootloader. You will notice that if you disconnect from the printer after setting a temperature and then reconnect, the printer will reset and start cooling down.

ntoff, 2017-10-05:
>The process of reconnecting to the serial bus of the controller resets the controller.

In this case OP was shutting down so I wouldn't expect anything would *re*connect to the board, *dis*connect, sure but reconnecting? I totally agree it's something hinkey with the serial reset Arduino's do but I don't think I've ever had mine do it on shutdown.

galamdring, 2017-10-05:
It does it on reboot when the USB bus is started up. Applying power to the
USB port causes the arduino to reset. Disconnecting the USB bus may cause a
reset as well, but I don't think so.

foosel, 2017-10-09:
A warning message has been implemented by the just merged #2146. Available on `maintenance`, soon on `devel`, will be part of 1.3.6.

ntoff, 2017-10-13:
I just thought of something, the auto updater will just automatically restart OctoPrint when it's done, won't that (be even more likely to) also trigger a reconnect? 

Maybe the auto updater should also have the warning applied to its dialog too? And a warning that it's going to auto restart OctoPrint once the update is done.

scottpost4, 2017-10-13:
That's a good point. Something warning the user that they should not perform an update/restart when the printer is actively printing is probably a good idea as well.

ntoff, 2017-10-13:
I mean, for the record, I'm pretty sure OctoPrint won't even let you update if OctoPrint is what started the print, but in this case we're talking about people who start a print via the LCD panel on the printer, so as far as OctoPrint knows, the printer is idle.

Maybe a future feature request? A way of detecting if the printer is idle? I know not all firmware issues the "wait" or "busy" thing but I guess at least that's a place to start.

scottpost4, 2017-10-13:
Even something as simple as a button in Octoprint that, when toggled on, tells the system "hey, I've started printing from the printer, please stay on to record/monitor settings/etc. and DON'T try to send any data or reset. And if I tell you to reset, please yell at me in the form of dialogue boxes."
I'm not suggesting this is the best way to implement this, but it's the first thing I thought of.

ntoff, 2017-10-13:
Too unreliable, people will still post issues like "I forgot to tick the box and octoprint restarted, why can't it be automatic?", people are actually doing that now with the timelapse functionality because they forget to enable it pre-print.

Either way: https://github.com/ntoff/OctoPrint/tree/improve/auto-update-warn-sd-interupt

[image]

How's this?

scottpost4, 2017-10-13:
Haha yes, I assumed as much. I knew it wasn't a very good implementation of that idea. It's rather unfortunate that the hardware we're dealing with causes us to create all of these workarounds. Then again, Octoprint is much better than any built-in interface on all of the machines I've seen so far.
That screenshot (once again) looks excellent. Were I a user who didn't know what I do now, both of these would have given me the opportunity I needed to cancel and wait for my print to finish.

scottpost4, 2017-10-13:
I also like that all of the proceed buttons in the screenshots are red. Training the user to interpret a red button as something that will interrupt the software and whatever actions it / they are performing.

foosel, 2017-10-13:
@ntoff send a PR :) Happy to also include this, definitely makes sense!

ntoff, 2017-10-13:
@scottpost4 I'm pretty sure the buttons are red by default, that's not something I've changed. Are your confirmation buttons not red?

scottpost4, 2017-10-13:
@ntoff apparently my glasses needed cleaning as well. Yes they are haha

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel