OctoPrint/OctoPrint #2132: rpi cam not starting

Penfold81, 2017-09-29:
Hi guys,

I have an issue with the new version of octoprint....

I have just installed on two different fresh installs on pi3's with the rpi cam 2.1. i have it all working fine and even got a screen working on one of them however i cannot get the webcam stream to come on? I keep getting webcam stream not loaded?

any ideas?

Dave

Penfold81, 2017-09-30:
after yet another fresh install it worked

closed 2017-09-30 by Penfold81

msbeck, 2018-06-02:
@Penfold81 can you share what your final stream url was? I updated past a handful of versions and my webcam simply stopped working as well.

[image]