OctoPrint/OctoPrint #2134: Octopi 1.4xx disconnecting

thomllama, 2017-10-01:
_#### What were you doing?_
Printing or not, the Pi/Octoprint would loose it's connection with the network. Was odd as sometimes the G-code animation would stay, but the camera image would be lost and any and all commands/buttons pushed would not be taken. Uploading a new g-code file would fail. Usually happens around 5 minutes of run time. Printing, not printing, doesn't matter.  

Have installed image on 3 different SD cards. plug or unplug the camera doesn't matter. When it disconnects the only way to communicate with it is to kill the power and restart it. 
Tried remedies - Power supply - Tried others but Original is WAY more than enough as it's from a Ipad and is the only one that I have that I can use the Pad and have it charge the pad at the same time, believe it's a 4.5 or 5A charger? 
Have uninstalled all plug ins, tried 3 different SD cards (4 gig, 8 gig and 16 gig) all of different sizes and brands

I have back stepped to Octopi install 1.3 and it is now running just fine .. sooo (not allowing the Octoprint update that popped up on start up) 

I_deally provide exact steps to follow in order to reproduce your problem:_

1. turn on Pi/printer.. do what ever 
2. Start a print or just let it sit for 5 minutes( could be more, could be less, but right around that time it always happens) 
3. try and to connect/view the Octopi/printer and it fails to show in the web browser. Sometimes the G-code animation runs, but terminal readouts stop and things like the camera go black. No command buttons work.  

_#### What did you expect to happen?_
commands to move/interact with Pi/Printer

_#### What happened instead?]_
Nothing at all usually

_#### Did the same happen when running OctoPrint in safe mode?_
Never tried Safe mode, honestly didn't know about it. Will try and report back.. 

_#### Branch & Commit or Version of OctoPrint_
Octopi install - octopi-jessie-lite-0.14.0 (Octoprint install 1.3.4 master branch)

_#### Operating System running OctoPrint_
Mac OSX Sierra and High Sierra (beta)
[OctoPi, Linux, Windows, MacOS, something else? With version please.
OctoPi's version can be found in /etc/octopi_version]
octopi-jessie-lite-0.14.0
_#### Printer model & used firmware incl. version_
Max Micron printer- Marlin 1.1.x

_#### Browser and Version of Browser, Operating System running Browser_
all browsers, Safari, Chrome, FireFox and also trying Brave (which is awesome by the way ;) )
[If applicable, always include if unsure.]

_#### Link to octoprint.log_
https://gist.github.com/thomllama/a644f5555714d8449b9810333a6a075f

I have read the FAQ. yes

thomllama, 2017-10-01:
other notes/info.. LOTS of ideas/help over on facebook in he Octoprint users group if yaz want to scan all that... https://www.facebook.com/groups/piprint/permalink/357234088048654/?comment_id=357378528034210&notif_id=1506892786469617&notif_t=group_comment

ntoff, 2017-10-02:
I take it you mean the wifi is disconnecting, and not the printer? This has zero to do with OctoPrint, all OctoPrint is responsible for is the serial connection to the printer, the operating system handles the wifi.

This is the bug tracker for Octo**Print**, which is only a single part of the pre-built OctoPi image. If you believe OctoPi is to blame, the OctoPi bug tracker is here: https://github.com/guysoft/OctoPi/

foosel, 2017-10-02:
As @ntoff already pointed out, this is not a bug in OctoPrint. It might be an issue with OctoPi, or with your network, and is a perfect example of a support question that should rather be asked in the official support channel (mailing list or g+ community) instead of using the bug tracker for this.

I'm closing this since there's nothing to be done in OctoPrint here. Comments are still possible however.

closed 2017-10-02 by foosel

thomllama, 2017-10-02:
right, sorry, I keep forgetting that Octoprint is basically just a program on the Pi, not the Operating System for it . :) .  SO I shouldn't have any issues updating Octoprint with the little window pops up.. thanks as always!!