OctoPrint/OctoPrint #2138: Serial Timeout

dtynan, 2017-10-03:
#### What were you doing?

I use a Mendel90 and have added the Nophead mod for turning on and off the printer from Octoprint. I'm running Octoprint on a RPI using Octopi. When I turn on the printer, the /dev/ttyUSB0 device shows up correctly, but Octoprint fails most of the time while trying to connect to the printer. It times out almost immediately:

`Connecting to: /dev/ttyUSB0
Error while connecting to /dev/ttyUSB0: 'Timeout'`

If I click on Connect again, I'll get the same message. If I keep trying it, eventually it will connect and initialise properly. Once connected, I never see any issues with data over the wire or the connection timing out thereafter. It's just on the initial handshake.

This appears with all versions of the code for the last year or so. Mostly it's just a nuisance as I can eventually connect but it can take 20 or 30 connection attempts, as well as the repeated dialog boxes on the screen.

#### What did you expect to happen?

I would expect that the time-out would take a few seconds rather than happening instantaneously. Or I would expect Octoprint to connect 

#### What happened instead?

Connection timeout.

#### Did the same happen when running OctoPrint in safe mode?

Generally if I restart Octoprint *after* I have turned on the printer, the connection is OK so when I bring up the system in Safe Mode, it works correctly. I will try turning off the printer within Safe Mode and then turning it back on again to see does the problem reoccur.

#### Branch & Commit or Version of OctoPrint

[Can be found in the lower left corner of the web interface. ALWAYS INCLUDE.]

#### Operating System running OctoPrint

OctoPi

#### Printer model & used firmware incl. version

Mendel90 with a Melzi board running Marlin v1.

#### Browser and Version of Browser, Operating System running Browser

Chrome on Ubuntu and Mac

#### Link to octoprint.log

`2017-10-03 16:27:16,686 - octoprint.util.comm - ERROR - Error while connecting to /dev/ttyUSB0: 'Timeout'
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/util/comm.py", line 1705, in _detectPort
    programmer.connect(p)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/util/avr_isp/stk500v2.py", line 34, in connect
    self.sendMessage([1])
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/util/avr_isp/stk500v2.py", line 108, in sendMessage
    return self.recvMessage()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/util/avr_isp/stk500v2.py", line 116, in recvMessage
    raise ispBase.IspError("Timeout")
IspError: 'Timeout'`

#### Link to contents of terminal tab or serial.log

Just the above message repeated every time I press Connect

#### Link to contents of Javascript console in the browser

#### Screenshot(s)/video(s) showing the problem:

I have read the FAQ.

As it's a specific problem to my printer, I'm happy to try and chase it down, but could do with some help as to what the code is trying to do at connect time. I am suspicious that perhaps there is some sort of cruddy serial data on the line and this is upsetting the serial I/O.

nophead, 2017-10-03:
I have connection timeout and autodetection timeout both set to 20s.

dtynan, 2017-10-03:
I have "Communication Timeout" set to 30s, "Connection Timeout" set to 20s, and "Autodetection Timeout" set to 30s. But it doesn't wait anywhere near that long. The timeout is all-but-instantaneous. I need to go into that STK500 code, but I'm hoping someone will go "**Oh I know what that is...**" before I do that.

foosel, 2017-10-03:
Try selecting the port to connect to instead of letting it auto detect. Auto detection doesn't work with all controllers. 1.3.5 will make that a bit better when there is only on port as option, but on 1.3.4 it still is a bit more stupid and always attempts to enter programmer mode to identify possible printers (an inheritance).

ntoff, 2017-10-03:
I found with one of my boards, disabling the serial reset line helped with connection timeouts. I think the bootloader must have sat around waiting for longer than normal or something.

I shoved a capacitor across the ground and reset pin like this person: http://forum.arduino.cc/index.php?topic=28167.msg208351#msg208351

jdevelop, 2017-10-04:
Got the same issue with the most recent version of OctoPrint on Arch Linux:

```
Connecting to: /dev/ttyACM0
Changing monitoring state from 'Offline' to 'Opening serial port'
Connected to: Serial<id=0x6ec49eb0, open=True>(port='/dev/ttyACM0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=40.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
Changing monitoring state from 'Opening serial port' to 'Connecting'
Send: N0 M110 N0*125
There was a timeout while trying to connect to the printer
Changing monitoring state from 'Connecting' to 'Offline'
Connection closed, closing down monitor
```

the config.yaml:

```
serial:
  baudrate: 115200
  port: /dev/ttyACM0
  timeout:
    communication: 40
    connection: 40
    detection: 0.5
    sdStatus: 1
    temperature: 5
    temperatureTargetSet: 2
```

logs:

```
2017-10-04 03:13:11,300 - octoprint.plugins.discovery - INFO - Registered OctoPrint instance on sender for SSDP
2017-10-04 03:13:11,301 - octoprint.server - INFO - Listening on http://0.0.0.0:5000
2017-10-04 03:13:11,496 - octoprint.plugins.pluginmanager - INFO - Loaded plugin repository data from disk, was still valid
2017-10-04 03:13:11,833 - octoprint.plugins.pluginmanager - INFO - Loaded notices from disk, was still valid
2017-10-04 03:13:12,151 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=en', 'path': '/', 'base_url': 'http://10.0.0.85:5000/'}
2017-10-04 03:13:30,205 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.169
2017-10-04 03:13:35,211 - octoprint.server.preemptive_cache - INFO - ... done in 23.06s
2017-10-04 03:13:35,741 - octoprint.server.util.sockjs - INFO - Client connection closed: 10.0.0.169
2017-10-04 03:13:36,815 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.0.169
2017-10-04 03:13:41,617 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
```

Anything else I could try out?

foosel, 2017-10-04:
> Got the same issue with the most recent version of OctoPrint on Arch Linux

Apologies, but you don't. The timeout reported by @dtynan happens in the stk500 submodule during autodetection of the port to use. Your's happens when attempting to perform a handshake with the printer, which never replies to the sent `M110` (it should say `ok` to that, or anything really, but instead just does nothing).

There's nothing really that OctoPrint can do if the printer doesn't respond to an `M110` sent on the correct printer port and at the correct baudrate. If your printer simply takes a bit too long to reset after the serial connection, it might help to further increase the connection timeout, but considering you already have it set to 40s I have my doubts here. 

Are you 100% certain `/dev/ttyACM0` is in fact your printer and not another serial device also connected to the machine in question? Because - assuming that 115200 is indeed the correct baudrate for this printer - this certainly doesn't look like it. Check the syslog after plugging in the printer to make sure. Try connecting with another host (e.g. Pronterface/printrun), see how that behaves.

dtynan, 2017-10-04:
So, doing a bit more digging (and in lieu of the 1.3.5 release which may obsolete this), I found that the following "fix" seems to resolve the issue:

```diff
--- a/src/octoprint/util/avr_isp/stk500v2.py
+++ b/src/octoprint/util/avr_isp/stk500v2.py
@@ -111,9 +111,13 @@ class Stk500v2(ispBase.IspBase):
                state = 'Start'
                checksum = 0
                while True:
-                       s = self.serial.read()
-                       if len(s) < 1:
+                       for attempt in range(0, 5):
+                               s = self.serial.read()
+                               if len(s) > 0:
+                                       break
+                       if attempt > 3:
                                raise ispBase.IspError("Timeout")
                        b = struct.unpack(">B", s)[0]
                        checksum ^= b
                        #print(hex(b))

```

It's not particularly clean, and to really drag out the kludge-ness of the fix, I should put a sleep in there after the break. But it seems to resolve the hiccup, whatever it is. The patch will resolve the issue for now, and it'll get blown away by the 1.3.5 release when that hits the streets, so I can try again then and see if the problem still exists in the new release.

foosel, 2017-10-04:
1.3.5 doesn't do any particular magic - it simply assumes the printer to be the only available serial port. So your patch certainly looks interesting, it would get around controllers that take a bit too long to enter the requested mode. I'd probably change it to

``` python
while True:
    for _ in range(0, 5):
        s = self.serial.read()
        if len(s) > 0:
             break
        time.sleep(0.1)
    else:
        raise ispBase.IspError("Timeout")
```

Still, in general I'd rather get rid of the "test if you can enter programmer mode" approach altogether and instead just iterate over the available serial ports. The only issue there is that if you also set the baudrate to auto that might take AGES if you have several not-printer serial devices connected to your system. Which probably is more the very rare exception than the rule of course, but still...

jdevelop, 2017-10-04:
@foosel perhaps you're right. I've tried to connect to `/dev/ttyACM0` via `screen` and `tio`, and got something strange - sometimes it works, sometimes it doesn't. So I turned the printer off for a while, and then tried again - now both `screen` and `tio` respond with `B15 ok`, and OctoPrint connects to the printer just fine.

Thanks!

foosel, 2017-10-11:
I just pushed the added delay to `maintenance` (and `devel`) so that it will be part of the 1.3.6 release. I'm not entirely happy with this since it basically introduces more lenient timeouts where there actually shouldn't be any, but since this code is only used for auto detection anyhow (and will probably sooner rather than later be made to go the way of the dodo) I'd say it's at least worth a test.

Gonzo1981, 2017-10-29:
i have a timeout problem too:
my Log with 1.3.4 : 
2017-10-29 16:29:08,215 - Connecting to: /dev/ttyUSB0
2017-10-29 16:29:08,227 - Changing monitoring state from 'Offline' to 'Opening serial port'
2017-10-29 16:29:08,229 - Connected to: Serial<id=0x6e2bbbb0, open=True>(port='/dev/ttyUSB0', baudrate=115200, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2017-10-29 16:29:08,230 - Changing monitoring state from 'Opening serial port' to 'Connecting'
2017-10-29 16:29:08,234 - Send: N0 M110 N0*125
2017-10-29 16:29:10,042 - Recv: start
2017-10-29 16:29:10,048 - Recv: echo:Marlin 1.1.6
2017-10-29 16:29:10,049 - Recv: 
2017-10-29 16:29:10,053 - Recv: echo: Last Updated: 2017-10-10 12:00 | Author: (Creality CR-10)
2017-10-29 16:29:10,056 - Recv: echo:Compiled: Oct 28 2017
2017-10-29 16:29:10,059 - Send: N0 M110 N0*125
2017-10-29 16:29:10,061 - Recv: echo: Free Memory: 1187  PlannerBufferBytes: 1232
2017-10-29 16:29:10,065 - Recv: echo:EEPROM version mismatch (EEPROM=? Marlin=V41)
2017-10-29 16:29:10,068 - Recv: echo:Hardcoded Default Settings Loaded
2017-10-29 16:29:13,078 - Recv: ok
2017-10-29 16:29:13,084 - Changing monitoring state from 'Connecting' to 'Operational'
2017-10-29 16:29:13,093 - Send: N0 M110 N0*125
2017-10-29 16:29:13,109 - Recv: ok
2017-10-29 16:29:13,111 - Send: N1 M115*39
2017-10-29 16:29:13,177 - Recv: FIRMWARE_NAME:Marlin 1.1.6 (Github) SOURCE_CODE_URL:https://github.com/MarlinFirmware/Marlin PROTOCOL_VERSION:1.0 MACHINE_TYPE:G 3D EXTRUDER_COUNT:1 UUID:cede2a2f-41a2-4748-9b12-c55c62f367ff
2017-10-29 16:29:13,180 - Recv: Cap:EEPROM:1
2017-10-29 16:29:13,182 - Recv: Cap:AUTOREPORT_TEMP:1
2017-10-29 16:29:13,183 - Recv: Cap:PROGRESS:0
2017-10-29 16:29:13,185 - Recv: Cap:PRINT_JOB:1
2017-10-29 16:29:13,187 - Recv: Cap:AUTOLEVEL:0
2017-10-29 16:29:13,189 - Recv: Cap:Z_PROBE:0
2017-10-29 16:29:13,190 - Recv: Cap:LEVELING_DATA:0
2017-10-29 16:29:13,192 - Recv: Cap:SOFTWARE_POWER:0
2017-10-29 16:29:13,194 - Recv: Cap:TOGGLE_LIGHTS:0
2017-10-29 16:29:13,197 - Recv: Cap:CASE_LIGHT_BRIGHTNESS:0
2017-10-29 16:29:13,199 - Recv: Cap:EMERGENCY_PARSER:0
2017-10-29 16:29:13,200 - Recv: ok
2017-10-29 16:29:13,201 - Send: M21
2017-10-29 16:29:13,216 - Recv: echo:SD card ok
2017-10-29 16:29:13,375 - Recv: ok
2017-10-29 16:29:13,377 - Send: M20
2017-10-29 16:29:13,392 - Recv: Begin file list
2017-10-29 16:29:13,757 - Recv: /ANLEIT~1/~1.MOD/CAT~1.GCO 20681570
2017-10-29 16:29:13,776 - Recv: /ANLEIT~1/~1.MOD/YAXISW~1/CR-10~1.GCO 2653383
2017-10-29 16:29:13,808 - Recv: /ANLEIT~1/CR-10_~1/LUOSIL~1.GCO 8563406
2017-10-29 16:29:13,871 - Recv: echo:Cannot open subdir g
2017-10-29 16:29:13,874 - Recv: echo:Cannot open subdir BJECT
2017-10-29 16:29:13,968 - Recv: echo:Cannot open subdir q%/���r��{
2017-10-29 16:29:13,983 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,046 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,049 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,118 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,122 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,181 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,184 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,251 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,255 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,301 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,315 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,386 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,390 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,434 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,449 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,518 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,522 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,567 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,582 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,646 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,649 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,718 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,722 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:29:14,781 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:29:14,784 - Recv: echo:Cannot open subdir m~m�B�*!.��!
[... 261 lines cut ...]
2017-10-29 16:35:32,921 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:35:32,924 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:35:32,993 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:35:32,996 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:35:33,055 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:35:33,059 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:35:33,126 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:35:33,130 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:35:33,174 - Recv: echo:Cannot open subdir �����WM�.���
2017-10-29 16:35:33,190 - Recv: echo:Cannot open subdir m~m�B�*!.��!
2017-10-29 16:35:33,483 - Recv: start
2017-10-29 16:35:33,484 - Printer sent 'start' while already operational. External reset? Resetting line numbers to be on the safe side
2017-10-29 16:35:33,487 - Recv: echo:Marlin 1.1.6
2017-10-29 16:35:33,489 - Recv: 
[... cut ...]
2017-10-29 16:35:33,506 - Recv: echo:Hardcoded Default Settings Loaded
2017-10-29 16:35:38,472 - Recv: echo:SD card ok
2017-10-29 16:36:40,348 - Communication timeout while idle, trying to trigger response from printer. Configure long running commands or increase communication timeout if that happens regularly on specific commands or long moves.
2017-10-29 16:36:40,354 - Send: N0 M110 N0*125
2017-10-29 16:36:40,361 - Recv: ok

i don't Receive Temperature and get a little bit later timeout failure

foosel, 2017-10-30:
@Gonzo1981 Your printer restarts right in the middle of things. That confuses OctoPrint (e.g. it never receives an ok or the end of file list at the very least for an earlier `M20`). I can see that I harden it further against printer issues like that in 1.3.6, but I strongly suggest you get to the ground of why your printer does a double-reset on connect, that is NOT normal, and neither is this flood of errors regarding files that on the SD card in your printer. My guess would be that the two issues are related and your Marlin resets while reading the card or something like that, so maybe try what happens when you do not have it in the printer on connect.

Gonzo1981, 2017-10-30:
Thank you very much. It was the SD Card. After format it works fine. Only the USB Camera don’t work now. Its a Microsoft Xbox 360 Live Cam.

ntoff, 2017-10-31:
Is the camera on the supported list? Not every camera is supported by raspbian / mjpeg streamer.

https://github.com/foosel/OctoPrint/wiki/Webcams-known-to-work Maybe see if it needs any extra parameters set in `/boot/octopi.txt`

buzztiaan, 2017-10-31:
the Xbox 360 Live Vision Camera is just a UVC camera ..   (the one on https://www.amazon.co.uk/Official-Microsoft-Xbox-Vision-Webcam/dp/B0054UMR8O )

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel