OctoPrint/OctoPrint #2143: [Request] Manipulate GCode from GCode Viewer

HeinerTholen, 2017-10-05:
Hey there,

I like to change filament and fiddle with the parts during prints. Currently, my solution is to sit next to the printer, wait, and hit pause in the right moment. It does the job.

However, the ideal feature would let me pick a solid line in the GCode viewer tab in OctoPrint and then give me a field to insert code before or after that line. Thereby I could retract, move the hotend away, beep, wait, play "Pour Elise", etc. 

Better even would be a feature that takes me from the solid line in the GCode viewer to the actual GCode line in a text editor, where I could make any modification. 

I guess it would be quite some effort, but it would be a legit feature, no? =)

Cheers and thank you for making OctoPrint! It's very very incredible!
Heiner

foosel, 2017-10-11:
> I guess it would be quite some effort, but it would be a legit feature, no?

Yes to both. Marked correspondingly ;)

tvi675, 2018-01-14:
To extend this it would be amazing if you could mark an area to ignore in the gcode viewer, for example if you’re printing multiple parts, one fails but the rest are ok, rather than waste a long print you can just ignore the rest of that part.

bradcfisher, 2018-07-06:
@tvi675 I've created a plugin that does exactly what you suggested.  There's an open PR to have it listed on the plugins site, but you can install it right before then using this url if you want to try it out:
https://github.com/bradcfisher/OctoPrint-ExcludeRegionPlugin/archive/0.1.0.zip

foosel, 2018-07-06:
@bradcfisher correction - the PR is no longer open but merged ;)

Closing this since there is now a plugin 🎉

foosel, 2018-07-06:
Oops... reopening since the original issue was about also inserting lines - my bad :)

LazeMSS, 2021-03-31:
Possible by plugin:
https://plugins.octoprint.org/plugins/GcodeEditor/

closed 2021-03-31 by LazeMSS