OctoPrint/OctoPrint #2144: Octoprint intertface and interaction extremly slow with Pi 2 model B

sn4k3, 2017-10-05:
I have one pi 3 with octoprint and it always work well, no problems or slow down.
But now i have setup other printer and i used my old pi 2 model B, i deploy the most recent octoprint to the SDCard and turn on the pi. It boot up normaly and ssh terminal are normal, the problem come when i try to use the octoprint interface. First it take ages to auto connect to printer, and only 10m to 15m after pi boot i'm able to access the interface. Send commands to printer have a huge delay, for example i click to move left  the X axis, after 5s it moves, sometimes even more time...
If i refresh the page (F5) it took minutes to be able to use it again.
I can also notice the ssh terminal process the comands slower when octoprint is still booting or refreshing.

My setup:

- Prusa MK2s
- Pi 2 model B with pi camera v1
- Wired Ethernet connection CAT 7 cable (The same i use with pi 3)
- Install image using the Etcher and octoprint image.
- Configure octoprint first dialog
- Run apt-get update & upgrade
- Use interface without any mods or other configs

Here the things i have done and tested:

- 3 different SDCards (1 of them is the same i use on pi3 and is a SDXC card, others are normal Class 10)
- Overclock pi by 100Mhz (i have heatshinks)
- Disable camera by kill the service
None of that fix my problem, kill the cam service help a bit but interface still a bit slow.

I also have done speed tests to internet, the results are:

> speedtest-cli
> Hosted by Vodafone PT (Lisbon) [69.45 km]: 57.289 ms
> Testing download speed................................................................................
> Download: 21.84 Mbit/s
> Testing upload speed................................................................................................
> Upload: 16.20 Mbit/s

When do same test on pi 3 i get +/- 25ms and 95 Mbit/s on both download and upload (My internet connection is 200/100) so the pi 3 can get the max bandwidth of it due no gigabite port. Note that i use same cable on pi2 and pi3 for testing.

Only by speed test i can notice something wrong, pi2 should be able to get same results as pi3.

When octoprint is booting the the process takes 20% of the CPU and camstream 20% as well.
After sometime beeing on the proccess are:

> pi@octopi:~ $ ps aux
> USER       PID %CPU %MEM    VSZ   RSS TTY      STAT START   TIME COMMAND
> root         1  0.1  1.0   5396  3820 ?        Ss   14:04   0:04 /sbin/init
> root         2  0.0  0.0      0     0 ?        S    14:04   0:00 [kthreadd]
> root         3  0.3  0.0      0     0 ?        S    14:04   0:08 [ksoftirqd/0]
> root         5  0.0  0.0      0     0 ?        S<   14:04   0:00 [kworker/0:0H]
> root         6  0.0  0.0      0     0 ?        S    14:04   0:00 [kworker/u2:0]
> root         7  0.0  0.0      0     0 ?        S<   14:04   0:00 [lru-add-drain]
> root         8  0.0  0.0      0     0 ?        S    14:04   0:00 [kdevtmpfs]
> root         9  0.0  0.0      0     0 ?        S<   14:04   0:00 [netns]
> root        10  0.0  0.0      0     0 ?        S    14:04   0:00 [khungtaskd]
> root        11  0.0  0.0      0     0 ?        S    14:04   0:00 [oom_reaper]
> root        12  0.0  0.0      0     0 ?        S<   14:04   0:00 [writeback]
> root        13  0.0  0.0      0     0 ?        S    14:04   0:00 [kcompactd0]
> root        14  0.0  0.0      0     0 ?        S<   14:04   0:00 [crypto]
> root        15  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        16  0.0  0.0      0     0 ?        S<   14:04   0:00 [kblockd]
> root        17  0.0  0.0      0     0 ?        S<   14:04   0:00 [watchdogd]
> root        19  0.0  0.0      0     0 ?        S<   14:04   0:00 [rpciod]
> root        20  0.0  0.0      0     0 ?        S<   14:04   0:00 [xprtiod]
> root        21  0.0  0.0      0     0 ?        S    14:04   0:00 [kswapd0]
> root        22  0.0  0.0      0     0 ?        S<   14:04   0:00 [nfsiod]
> root        32  0.0  0.0      0     0 ?        S<   14:04   0:00 [kthrotld]
> root        33  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        34  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        35  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        36  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        37  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        38  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        39  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        40  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
> root        41  0.0  0.0      0     0 ?        S<   14:04   0:00 [bioset]
[... 37 lines cut ...]
> root       366  0.0  0.0      0     0 ?        S<   14:04   0:00 [cfg80211]
> nobody     421  0.0  0.3   2276  1456 ?        Ss   14:04   0:00 /usr/sbin/thd --daemon --triggers /etc/triggerhappy/trigg
> pi         428 11.4 11.6 173356 44240 ?        Sl   14:04   4:05 /home/pi/oprint/bin/python /home/pi/oprint/bin/octoprint
> pi         434  0.0  0.7   5112  2932 ?        S    14:04   0:00 /bin/bash /root/bin/webcamd
> pi         462 11.0  0.7  90768  2812 ?        Sl   14:04   3:55 ./mjpg_streamer -o output_http.so -w ./www-octopi -n -i i
> root       622  0.0  0.4   2548  1644 ?        Ss   14:05   0:01 /sbin/dhcpcd -q -w
> root       624  0.0  1.1   7852  4292 ?        Ss   14:05   0:00 /usr/sbin/sshd -D
> root       635  0.0  0.1   1668   728 ?        Ss   14:05   0:00 /usr/sbin/haproxy-systemd-wrapper -f /etc/haproxy/haproxy
> haproxy    637  0.0  2.1  10216  8304 ?        S    14:05   0:00 /usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -p /run/hap
> haproxy    649  1.4  1.9  10760  7264 ?        Ss   14:05   0:31 /usr/sbin/haproxy -f /etc/haproxy/haproxy.cfg -p /run/hap
> ntp        652  0.0  0.9   5760  3684 ?        Ss   14:05   0:00 /usr/sbin/ntpd -p /var/run/ntpd.pid -g -u 106:111
> root       658  0.0  0.4   4036  1748 tty1     Ss+  14:05   0:00 /sbin/agetty --noclear tty1 linux
> root       659  0.0  0.5   3856  2028 ?        Ss+  14:05   0:00 /sbin/agetty --keep-baud 115200 38400 9600 ttyAMA0 vt102
> root       669  0.0  0.0      0     0 ?        S<   14:05   0:00 [kworker/0:1H]
> root       687  0.0  1.3  11540  4956 ?        Ss   14:06   0:00 sshd: pi [priv]
> root       705  0.0  1.2  11540  4896 ?        Ss   14:06   0:00 sshd: pi [priv]
> pi         709  0.1  1.0  11540  4144 ?        R    14:06   0:02 sshd: pi@pts/0
> pi         715  0.0  1.0  11540  4016 ?        S    14:06   0:00 sshd: pi@notty
> pi         717  0.0  0.4   2320  1580 ?        Ss   14:06   0:00 /usr/lib/openssh/sftp-server
> pi         718  0.0  1.2   6788  4796 pts/0    Ss   14:06   0:01 -bash
> root       746  0.0  0.0      0     0 ?        S    14:10   0:01 [kworker/0:0]
> root       787  0.0  0.0      0     0 ?        S    14:29   0:00 [kworker/0:1]
> pi        1347  0.0  0.5   4724  2164 pts/0    R+   14:40   0:00 ps aux

Regarding printing i don't have timeout or delay, but i lost almost 20m or more to start print something from a system turn on to print action

foosel, 2017-10-10:
> Octoprint official image

So OctoPi then I guess? What version? Looks like 0.14, but better safe than sorry.

Based on the information you've provided so far (web page taking long to load, clicks taking ages, but prints running fine), it sounds like the network in general is a problem here (and no, I don't know why either). If you hadn't stated that your prints run fine I'd also guessed that your Pi2 might get overloaded, but that doesn't fit the bill here. As a side note, a Pi2 in general should not have any issue with running OctoPrint and OctoPi (my main printer is still sporting a Pi2, so I'm 100% positive on that).

Just to make sure, check `top` *after* initial boot. From your `ps` output it doesn't look anything out of the ordinary is running, but something might take a lot more CPU load than it should (a CPU peak on startup is *normal* but it shouldn't persist). Your syslog might also be interesting, maybe something is going nuts in there. Also check what happens if you plug the SD from the Pi3 into the Pi2 to rule out any issues with the Pi2 itself. Finally, since you haven't said anything about that, test the PSU of your Pi3 on the Pi2, see if that improves things - insufficient power can cause all kinds of problems on Pis, including connection resets.

In general, for issues like that please first get in touch on the Mailinglist or the G+ community - it sounds more like an issue with your particular setup than a bug, and if it is a bug it is most likely not a bug in OctoPrint itself but rather the underlying OctoPi. This bug tracker is about OctoPrint, not OctoPi, and it is not a support forum. And when you *do* report a bug in OctoPrint, provide *all the requested information*, especially the logs. They are asked for for a reason, you are sabotaging yourself by not providing that info.

foosel, 2017-10-11:
If the very same software snapshot that works fine on your Pi3 shows the same kind of broken behaviour on your Pi2 then it is not an issue with the software but rather the hardware. I'd suggest replacing the Pi2. In any case, not a bug in OctoPrint but rather an environment issue. Closing.

closed 2017-10-11 by foosel

biscuitlad, 2018-09-01:
I have exactly this issue with Pi2 as well. It was working fine, I upgraded to the latest, both octoprint and octopi, and now it takes 5-10mins to for the server to start? What has broken with the new versions?

psy0rz, 2018-09-04:
I have this on the pi zero w.

dimkin-eu, 2018-10-06:
slow sd card ?

psy0rz, 2018-10-08:
I once checked the logs while it starts, and it was the precompiling of python stuff that took all the time. Also it was cpu bound.

SR-G, 2019-01-07:
Exactly the same problem on PINE64 with ARMBIAN (built on debian jessie, only armbian build available...). It's so slow that it's totally unusable, whereas the board is nearly doing nothing (99% idle).

CRCinAU, 2019-03-26:
I just downloaded the image of OctoPi 0.16.0 and put it on a Pi 3B and also thought it was extremely slow.

I'm currently doing an 'apt-get update && apt-get -y upgrade' and will try again.

foosel, 2019-03-26:
[image]

OctoPrint 1.3.10 on OctoPi 0.16.0 on an RPi3, connected via WiFi. Fast.

CRCinAU, 2019-03-26:
This is what I see:
[image]

Pinging the router over the wifi connection:
```
root@octopi:~# ping 10.1.1.254
PING 10.1.1.254 (10.1.1.254) 56(84) bytes of data.
64 bytes from 10.1.1.254: icmp_seq=1 ttl=64 time=2.81 ms
64 bytes from 10.1.1.254: icmp_seq=2 ttl=64 time=5.75 ms
64 bytes from 10.1.1.254: icmp_seq=3 ttl=64 time=4.63 ms
64 bytes from 10.1.1.254: icmp_seq=4 ttl=64 time=18.8 ms
64 bytes from 10.1.1.254: icmp_seq=5 ttl=64 time=15.0 ms
^C
--- 10.1.1.254 ping statistics ---
5 packets transmitted, 5 received, 0% packet loss, time 4005ms
rtt min/avg/max/mdev = 2.812/9.418/18.832/6.329 ms
```

From top, I notice:
```
top - 11:24:12 up 7 min,  1 user,  load average: 0.28, 0.40, 0.25
Tasks: 103 total,   2 running,  59 sleeping,   0 stopped,   0 zombie
%Cpu(s): 25.1 us,  1.2 sy,  0.0 ni, 73.8 id,  0.0 wa,  0.0 hi,  0.0 si,  0.0 st
KiB Mem :   896668 total,   691484 free,    87468 used,   117716 buff/cache
KiB Swap:   102396 total,   102396 free,        0 used.   751436 avail Mem 

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                                                                                                                  
  440 pi        20   0  214240  55252   8940 R 105.6  6.2   2:19.69 /home/pi/oprint/bin/python2 /home/pi/oprint/bin/octoprint serve --host=127.0.0.1 --port=5000                                                                             
  946 root      20   0    8260   3352   2796 R   1.0  0.4   0:00.61 top                                                                                                                                                                      
  190 root      20   0       0      0      0 S   0.3  0.0   0:00.35 [ft5406]                                                                                                                                                                 
  792 root      20   0       0      0      0 I   0.3  0.0   0:00.64 [kworker/u8:3]
```

CRCinAU, 2019-03-26:
I wonder, is this possibly because I'm not connected to a 3D printer yet? I've set up a profile, but I haven't received my printer yet - hence it can't connect to one.

foosel, 2019-03-29:
> I wonder, is this possibly because I'm not connected to a 3D printer yet?

No. My example was also from an instance to which no printer was connected.

Based on your network screenshot, loading the initial HTML takes a very long time, and loading a completely static file does as well. Unless my math is wrong, the packed_libs.js file was received by your browser with less than 200kb/s transmission speed. That seems very slow for a local network. Your screenshot also shows way longer green = receiving durations. Also for you the 1.14MB packed_libs.js file takes over 8s and mine only 1s (and even less when I don't intentionally nuke the cache). As mentioned, my measurement was done on a Pi3 via WiFi as well.

I'd start by running in safe mode. If the issue is gone then it's some third party plugin causing problems. Then I'd continue with debugging the network speed between your pi and your client machine.

CRCinAU, 2019-03-30:
Interestingly, I swapped the SD card from a Pi3 to a Pi3+ and the problem has gone away....

I'm doing some prints at the moment, but I'll see if I can get some time to swap the card a few more times and see if I can figure out what is going on...