OctoPrint/OctoPrint #2147: Reordering of UI items causes misalignment of temp graph legend

ntoff, 2017-10-06:
#### What were you doing?

Reordering the tabs and sidebar items to my liking.

1. Start OctoPrint with `octoprint serve -b testhere` (or is that `octoprint -b testhere serve`? both work) or start from a fresh install (happens on an established install too but I wanted to be sure the config was clean)
2. Enable virtual printer, baud 250000, save connection info, connect on start up (not tested with an actual printer because one isn't attached but I would assume the same result)
3. Reorder the sidebar items and the tabs

config.yaml lies here (secrets redacted, not that it matters, it's a virtual install inside a virtual machine):  https://gist.github.com/ntoff/0b3f42a5b3e41168d19f26ec9d15a83e

#### What did you expect to happen?

Legend to remain aligned

#### What happened instead?

Not sure if it's the legend or the box border around the graph, the border itself seems to be the thing that moves around, either way, they meet up somewhere in the middle once the graph starts to receive temperature info. If no temperature info is received the graph looks more normal, though still not quite the same as it does in the default tab order.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5rc4 (rc/maintenance branch)

#### Operating System running OctoPrint

Windows 7 
linux mint 18 (I think) running in a vm on windows 10

#### Printer model & used firmware incl. version

built in virtual printer

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 61.0.3163.100 (Official Build) (64-bit) - windows 10
Chrome on android v57.0.2987.132
Firefox on linux mint 55.0.2 (64-bit)

#### Link to octoprint.log

https://gist.github.com/ntoff/bbde51babcfe334345b9b87eb925a992

#### Link to contents of terminal tab or serial.log

Not serial / printer related that I'm aware of, though the issue does seem to require the temperature graph receive temperature info to fully manifest so a connection to *something* is required, virtual printer or a real one.

#### Link to contents of Javascript console in the browser

Octoprint server on windows 7
[image]

OctoPrint server on linux mint in a VM
[image]

#### Screenshot(s)/video(s) showing the problem:

Mint VM on the left, Windows7 on the right, both accessed from Chrome on a windows 10 PC

[image]

I have read the FAQ.

foosel, 2017-10-06:
Yeah, that's a regression (checked 1.3.4 and that doesn't show this). Considering it's purely of an aesthetic nature I think that it can wait to be fixed until 1.3.6 though. At least if there's nothing else that pops up in 1.3.5rc4.

foosel, 2017-10-09:
Found and fixed on `maintenance`, soon `devel`, will be part of 1.3.6.

commit b89b6970 (2017-10-09): Fix temperature plot init if it's not the 1st tab

If the temperature graph was not on the first tab, the plot would be
initialized without being visible, causing some sizing issues and
alignment issues with the Y axis ticks.

Forcing the plot to only get initialized once visible solves these
problems.

Fixes #2147
closed 2017-12-12 by foosel