OctoPrint/OctoPrint #2148: [Request] Support serial port rates 500000 and 1000000

fiveangle, 2017-10-08:
It's just a simple addition in comm.py:

```
 def baudrateList():
 	# sorted by likelihood
-	candidates = [115200, 250000, 230400, 57600, 38400, 19200, 9600]
+	candidates = [115200, 250000, 230400, 57600, 38400, 19200, 9600, 500000, 1000000]

```
I see a very small improvement on AT90USB Printrboard RevF4 when doing SD transfers (not a lot, but any extra bits are better than nothing :)

FYI, there is work happening on Marlin to improve SD bulk transfers, including XON/XOFF flow control (of course will be useless w/o host-side changes eventually): https://github.com/MarlinFirmware/Marlin/pull/7818

ntoff, 2017-10-08:
If you look in the serial options, you can add your own baud rates. Do you also see any side effects while printing? Increased error rates? Would the USB cable need to be higher quality? Do prints with lots of small segments run any smoother?

fiveangle, 2017-10-08:
Yes, it just seemed like these should be there by default since for any of the USB-native AVR units, the speed works well and offers approx 10% boost in SD transfer speed (1M vs 250k)

I've not observed any negative side effects running probably 6hrs across 3 prints using relatively high 120mm/s w/1000mm/s/s acc across a 6" square bed with Linear Advance enabled.  I will keep moniotring.

As this is just one step toward the more valuable implementation of XON/XOFF host-side flow control, we can close and we deal with them both together, assuming XON/XOFF is tackled at some point *shrug*.

Grogyan, 2017-10-08:
Please keep in mind that the higher baud, the higher the chance for lost signals.
General rule of thumb is 250K just happens to be in phase with a 16MHz oscillator.

If signals are lost in that, then it would be prudent to find the source of the EMI

fiveangle, 2017-10-09:
Native USB is not generally susceptible to the same type of "lost signals" issues as normal serial

foosel, 2017-10-09:
How many printers are there that actually support these baud rates (as in, do actually run native USB instead of the more common low speed serial converters)? 

A lot of users think "higher is better" and don't understand that it's their printer that has the final word on whether it supports a certain baudrate or not. So by adding these quite non-standard baudrates into the default list, I fear we'd see a stream of users selecting them and then opening tickets because "OctoPrint doesn't connect to my printer, fix it". This already happens way too often with 250000 baud (which is fairly uncommon supported out of the box as far as I can tell from the support requests I'm seeing). The idea so far is, if a power user needs different/higher baudrates, they can easily add it themselves.

ntoff, 2017-10-09:
well according to this: http://reprap.org/wiki/Taurino#Taurino_Power my board supports up to 2mbit. 

>ATmega16U2 (with enhanced firmware) for high speed USB serial connection (up to 2MBit)

this could be ... interesting -_- 

I wonder if repetier firmware works at higher than 250kbps

nophead, 2017-10-09:
I don't think baud rate means anything with native USB support. There is no
serial link other than the USB itself that runs at 12Mbits regardless.

ATmegas will certainly do at least 1Mbit serial hardware wise. I don't
think it would make much difference with OctoPrint though because it is the
Python comms that is the bottleneck at 250Kb.

ntoff, 2017-10-09:
well apparently my printer connected at 1000000, though I have no real idea how to benchmark it, I don't have any fancy logic analysers or anything. I can't transfer a file over because I don't think the SD reader in that printer works.

>because it is the Python comms that is the bottleneck at 250Kb.

So does that mean even something like pronterface wouldn't be able to keep up? Seeing as it's also python based.

nophead, 2017-10-09:
IIRC a stand alone Python program is about twice as fast as OctoPrint because OctoPrint is a lot more complicated with logging, multiple protocol handling and plugin hooks, etc.

JustAnother1, 2017-10-09:
> Native USB simply ignoring the baud rate was a good hint though - ... - what's up here?

Maybe the host respects the Baud rate ? 
If the Host limits the amount of bytes it tries to send to the client in a certain time (because it thinks that the client has, due to the low baud rate, no way of consuming that amount of bytes) than the higher baud rate would make a difference. 

Just guessing here. I never did any measurements.

But if you can choose, then please use native USB over UART every time!

foosel, 2017-10-10:
> But if you can choose, then please use native USB over UART every time!

I'd be surprised if pyserial allows me that much flexibility, but any hints are welcome.

JustAnother1, 2017-10-10:
> >   But if you can choose, then please use native USB over UART every time!

> I'd be surprised if pyserial allows me that much flexibility, but any hints are welcome.

I was unclear, sorry. The recommendation was intended for the Firmware / board selection. If someone needs to find the best board for a new printer, then I would recommend to go with the one that has native USB and not some USB to UART chip.

Also with host I refereed to the Operating system. I'm sure that Octoprint does not do any rate limiting.

After causing so much confusion the only hint I can give is that it is sometimes possible to get more information about the driver that "creates" the serial port. With this information Octoprint could try different strategies with different "drivers". Finding a USB Vendor ID of FTDI  or Silicon Labs is a strong indication that some USB to UART conversion is happening. Finding a VendorID/ProductID that is used by a native USB of a special firmware is a strong indication in the other direction.

On Linux (RasPi,..) lsusb gives detailed informations. also /dev/serial/by-id/ gives informations like that. I don't know if or how this information is available on other systems.

fiveangle, 2017-10-16:
As the intent behind Octoprint is for it to perform the operations, and that manual serial speeds can be entered now, closing.  If/when there is interest in supporting XON/XOFF flow controll for SD card transfer, I'm sure the topic will be revisited.

closed 2017-10-16 by fiveangle