OctoPrint/OctoPrint #2157: SockJS usage caused start failure

skrewz, 2017-10-18:
#### What were you doing?

Updated Octoprint 1.3.4 -> 1.3.5 using the software updater plugin.

#### What did you expect to happen?

Was expecting to see it coming up after the upgrade.

#### What happened instead?
`~/.octoprint/logs/octoprint.log`:
```
2017-10-18 15:34:47,266 - octoprint.cli.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/bin/octoprint", line 9, in <module>
    load_entry_point('OctoPrint==1.3.5', 'console_scripts', 'octoprint')()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/__init__.py", line 406, in main
    octo(args=args, prog_name="octoprint", auto_envvar_prefix="OCTOPRINT")
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 716, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 696, in main
    rv = self.invoke(ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 1037, in invoke
    return Command.invoke(self, ctx)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 889, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/cli/__init__.py", line 174, in octo
    ctx.invoke(serve_command, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/core.py", line 534, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/click-6.2-py2.7.egg/click/decorators.py", line 17, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/cli/server.py", line 150, in serve_command
    allow_root, logging, verbosity, safe_mode)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/cli/server.py", line 96, in run_server
    octoprint_server.run()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/__init__.py", line 460, in run
    session_kls=util.sockjs.ThreadSafeSession)
TypeError: __init__() got an unexpected keyword argument 'session_kls'
```

#### Did the same happen when running OctoPrint in safe mode?

Did not seem necessary to test.

#### Branch & Commit or Version of OctoPrint

`OctoPrint 1.3.5 (master branch)`

#### Operating System running OctoPrint

OctoPi 0.12.0.

#### Printer model & used firmware incl. version

Irrelevant.

#### Browser and Version of Browser, Operating System running Browser

Irrelevant.

#### Link to octoprint.log

https://gist.github.com/skrewz/b4ec7cc35f0d055e6b210e2dbdafafa5

#### Link to contents of terminal tab or serial.log

Irrelevant.

#### Link to contents of Javascript console in the browser

Irrelevant.

#### Screenshot(s)/video(s) showing the problem:

Irrelevant. Admittedly would've added some colour to these text snippets, though.

I have read the FAQ.

skrewz, 2017-10-18:
Please close with extreme prejudice.

I'll add that I went and ninja-hacked `/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/__init__.py` (from where the exception arose) to read:

```
                self._router = SockJSRouter(self._create_socket_connection, "/sockjs",
                                            dict(session_kls=util.sockjs.ThreadSafeSession))
```
as per what I can tell one should from https://pypi.python.org/pypi/sockjs-tornado. This makes Octoprint start as per usual, and I can wrap that up into a PR if such would be useful.

My python-fu is weak, however, and I can't tell if this should have been implicitly wrapped into a dict. And this beckons the question of why only I'm experiencing this---have I perhaps had an awkward upgrade path leading me to use a wrongly-configured python venv or somesuch?

So, with these caveats in mind: please close with extreme prejudice. Just couldn't let this sit in silence.

foosel, 2017-10-19:
This sounds rather like for some reason during your update your version of the `sockjs-tornado` dependency wasn't updated to 1.0.3 (which introduced `session_kls`). It should have been updated though, that's what #2136 was about and that was fixed in 1.3.5rc4.

What does `pip freeze | grep sockjs` say in your virtual environment? It should say 1.0.3. If it says 1.0.2 that's the problem and `pip install sockjs-tornado==1.0.3` should fix it. However, that should also have been done by the update script so I have no idea why you ran into this issue. Any errors during upgrade? Could you provide `plugin_softwareupdate_console.log` (should be right next to `octoprint.log`)?

skrewz, 2017-10-19:
Seen. Will post later today. :+1:

skrewz, 2017-10-19:
```
pi@octopi 15:14 ~ 0 source ~/oprint/bin/activate
pi@octopi 15:14 ~ 0 pip freeze -l | grep sockjs
sockjs-tornado==1.0.3
pi@octopi 15:14 ~ 0
```

`plugin_softwareupdate_console.log`: https://gist.github.com/skrewz/e34d736844ae4c99461ec1be6f2ae9ee

foosel, 2017-10-20:
Huh... Your update log indicates that the version was upgraded just fine and that also mirrors what your `pip freeze` output produces. 

But for some reason your OctoPrint isn't loading a `SockJSRouter` class that actually does support `session_kls` which version 1.0.3 [does](https://github.com/mrjoes/sockjs-tornado/commit/54290da5ea905f8ded44a6d8f728d4714b800967). To rule out any shenanigans caused by the somewhat ancient pip and python versions that are still present on a regular OctoPi 0.12 I also flashed that and did upgrade to 1.3.5 and I cannot reproduce this issue, runs fine.

Since I'm otherwise drawing a complete blank right now what's up there, let's do a little experiment. In your venv, fire up the interactive python shell and have it check on how the `__init__` function of whatever it loads as `SockJSRouter` looks like:

```
pi@octopi ~ $ source ~/oprint/bin/activate
(oprint)pi@octopi ~ $ python
Python 2.7.3 (default, Mar 18 2014, 05:13:23)
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import sockjs.tornado
>>> sockjs.tornado.SockJSRouter.__init__.__code__
<code object __init__ at 0x76ab1188, file "/home/pi/oprint/local/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg/sockjs/tornado/router.py", line 68>
>>> sockjs.tornado.SockJSRouter.__init__.__code__.co_varnames
('self', 'connection', 'prefix', 'user_settings', 'io_loop', 'session_kls', 'check_interval', 'base', 'p', 'k', 'v')
```

foosel, 2017-10-20:
Looks like sometime in the past you either installed OctoPrint globally instead of in the virtualenv, and your virtual environment chose that older library in your global dist-packages folder over the one installed locally. I think on OctoPi 0.12 the virtual environment was still configured to pick up dependencies from the system as well instead of being fully self contained, so that would explain it. 

A simple `sudo pip uninstall sockjs-tornado` should probably solve that, but what might be better is to switch your virtualenv to "no site packages" mode. As far as I understand, `touch ~/oprint/lib/python2.7/no-global-site-packages.txt` should already do that, but YMMV. To be on the safe side, you probably want to run `python setup.py install` again after that on the OctoPrint folder.

skrewz, 2017-10-20:
Indeed, an even better solution. I'll have a try.

Thanks for the courteous interest and support! :+1:

closed 2017-10-20 by skrewz

skrewz, 2017-10-20:
```
pi@octopi 14:14 ~ 0 which python
/usr/bin/python
pi@octopi 14:14 ~ 0 source ~/oprint/bin/activate
pi@octopi 14:14 ~ 0 which python
/home/pi/oprint/bin/python
pi@octopi 14:14 ~ 0 python mintest.py 
<code object __init__ at 0x76d17f98, file "/usr/local/lib/python2.7/dist-packages/sockjs/tornado/router.py", line 68>
('self', 'connection', 'prefix', 'user_settings', 'io_loop', 'check_interval', 'base', 'p', 'k', 'v')
pi@octopi 14:14 ~ 0 deactivate
pi@octopi 14:14 ~ 0 which python
/usr/bin/python
pi@octopi 14:14 ~ 0 touch ~/oprint/lib/python2.7/no-global-site-packages.txt
pi@octopi 14:14 ~ 0 source ~/oprint/bin/activate
pi@octopi 14:15 ~ 0 python mintest.py 
<code object __init__ at 0x76b681d0, file "/home/pi/oprint/local/lib/python2.7/site-packages/sockjs_tornado-1.0.3-py2.7.egg/sockjs/tornado/router.py", line 68>
('self', 'connection', 'prefix', 'user_settings', 'io_loop', 'session_kls', 'check_interval', 'base', 'p', 'k', 'v')
pi@octopi 14:15 ~ 0 cat mintest.py 
import sockjs.tornado
print sockjs.tornado.SockJSRouter.__init__.__code__
print sockjs.tornado.SockJSRouter.__init__.__code__.co_varnames
pi@octopi 14:15 ~ 0
```

---FYI. Doesn't get much clearer than that. And I've been meaning to find such a feature.

Running `python setup.py install` also took nicely in this venv.

foosel, 2017-10-20:
Ah, perfect! And I also learnt something new today, didn't know about the `no-global-site-packages.txt` thing until I took a quick look into the virtualenv sources to give you an easier option besides "redo the virtualenv".