OctoPrint/OctoPrint #2158: OctoPrint 1.3.4->1.3.5 update from GUI fails unless you have libyaml-dev installed

jarrodasmith, 2017-10-19:
If you use the current OctoPi image to bring up a new OctoPrint 1.3.4 system, you will be prompted to update to 1.3.5 from inside the server the first time you connect.  This update fails if you do not first manually install the libyaml-dev package in the underlying OS.

Tiefschneegiftler, 2017-10-24:
I have the same problem, however, mine comes from the OctoPi image 0.14 which unpacks to OctoPi 1.3.4 and than fails at updating to 1.3.5.
The problem is resulting from trying to update from the 1.3.4, as from the freshly installed image, to the 1.3.5 version via the notification in the start screen as well as when trying to update it via the settings.
Log: https://gist.github.com/Tiefschneegiftler/ecdafdbb99685551342063984f8b2f70

Is his thread related to: https://github.com/guysoft/OctoPi/issues/427 ?

foosel, 2017-10-24:
I don't know if it's related, I cannot reproduce either. Release procedure involves multiple update tests from various OctoPi and OctoPrint version states (incl. OctoPi 0.14 and OctoPrint 1.3.4 of course), if this was a general problem I would have noticed during those tests (or the four release candidates that went before the stable release). So it must be something odd.

What might help is `plugin_softwareupdate_console.log` from a failed attempt, should also be under `~/.octoprint/logs` on OctoPi (or Linux in general for that matter).

In general though there was no change at all in the yaml dependencies between 1.3.4 and 1.3.5.

foosel, 2017-10-24:
@jarrodasmith if you want this solved you should also provide some further information. I run several stock OctoPi 0.14 images here right now that updated fine without any manual installation steps.

jarrodasmith, 2017-10-24:
Thanks for looking at this.  I didn't provide more info because it was very easy to reproduce (also easy to work around) and I did so twice.  I'm surprised to hear you couldn't.  I will see if I can reproduce it a third time.

I don't have the logfile, Not knowing what would happen after the botched installation, I just reformatted that SD card and applied the workaround which is now in production (very nice, too!).

Jarrod

foosel, 2017-10-24:
@jarrodasmith Never assume that just because you can reproduce something easily, everyone can ;) And in case of OctoPrint you can bet on me and all of the people who tested the RCs to have run through the update process multiple times, so if it's failing for you, it must be something hard to reproduce that needs a lot of information to be debugged :)

jarrodasmith, 2017-10-24:
So in thinking back on this, "fails" may have been too strong a word.  I definitely saw an error in the log that some step had clearly failed due to missing yaml libs or headers.  I actually did not follow through to see if the software worked or not.

I will go through this again and submit the log file for your perusal, no assumptions on my part this time ;-)

Thanks again.

jarrod

foosel, 2017-10-24:
Ah... I think I know what you might have seen. If so that's actually a default "behaviour" of pyyaml. It attempts to compile with the headers, but if it doesn't find them, it still installs just fine (just without a native extension). It's always done that and there's nothing I can do about that, plus also no reason since what looks like an error is more like a warning.

If that indeed was what you were running into then whatever @Tiefschneegiftler encountered is something completely different. So let's see some logs from both of you and take it from there.

hazetekllc, 2017-10-24:
I have had the same issue three times now with fresh installs of 1.3.4

`2017-06-21 10:02:39,907   /home/pi/oprint/bin/python "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py" --branch= --force=false "/home/pi/OctoPrint" 1.3.5
2017-06-21 10:03:11,491 ! From https://github.com/foosel/OctoPrint
2017-06-21 10:03:11,493 ! fe481e1..b6cf895  master     -> origin/master
2017-06-21 10:03:11,494 ! * [new branch]      dev/slicingStlTransform -> origin/dev/slicingStlTransform
2017-06-21 10:03:11,495 ! 24dc120..880124b  devel      -> origin/devel
2017-06-21 10:03:11,496 ! 2aa7486..b94e513  fix/betterPluginDependencyTracking -> origin/fix/betterPluginDependencyTracking
2017-06-21 10:03:11,497 ! 086bd61..9a9e829  gh-pages   -> origin/gh-pages
2017-06-21 10:03:11,498 ! + dd8fb93...ece5edd improve/serverSideSessions -> origin/improve/serverSideSessions  (forced update)
2017-06-21 10:03:11,499 ! 4665070..38c410f  maintenance -> origin/maintenance
2017-06-21 10:03:11,500 ! fe481e1..77753ca  rc/devel   -> origin/rc/devel
2017-06-21 10:03:11,501 ! fe481e1..77753ca  rc/maintenance -> origin/rc/maintenance
2017-06-21 10:03:11,502 ! * [new branch]      regressionfix/2035 -> origin/regressionfix/2035
2017-06-21 10:03:11,503 ! 2b96e57..13de4b4  staging/maintenance -> origin/staging/maintenance
2017-06-21 10:03:11,504 ! * [new branch]      test/cancelWithoutM114M400 -> origin/test/cancelWithoutM114M400
2017-06-21 10:03:11,505 ! * [new branch]      test/issue1946_decouple_print_logging -> origin/test/issue1946_decouple_print_logging
2017-06-21 10:03:11,506 ! * [new tag]         1.3.5      -> 1.3.5
2017-06-21 10:03:11,506 ! * [new tag]         1.3.5rc1   -> 1.3.5rc1
2017-06-21 10:03:11,507 ! * [new tag]         1.3.5rc2   -> 1.3.5rc2
2017-06-21 10:03:11,508 ! * [new tag]         1.3.5rc3   -> 1.3.5rc3
2017-06-21 10:03:11,509 ! * [new tag]         1.3.5rc4   -> 1.3.5rc4
2017-06-21 10:03:37,297 > Python executable: '/home/pi/oprint/bin/python'
2017-06-21 10:03:37,299 > >>> Running: git diff --shortstat
2017-06-21 10:03:37,300 > > git diff --shortstat
2017-06-21 10:03:37,301 > >>> Running: git fetch
2017-06-21 10:03:37,301 > > git fetch
2017-06-21 10:03:37,302 > >>> Running: git pull
2017-06-21 10:03:37,303 > > git pull
2017-06-21 10:03:37,307 ! fatal: unable to access 'https://github.com/foosel/OctoPrint.git/': Could not resolve host: github.com
2017-06-21 10:03:37,309 ! Traceback (most recent call last):
2017-06-21 10:03:37,310 ! File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>
2017-06-21 10:03:37,311 ! main()
2017-06-21 10:03:37,312 ! File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 309, in main
2017-06-21 10:03:37,313 ! update_source(git_executable, folder, args.target, force=args.force, branch=args.branch)
2017-06-21 10:03:37,313 ! File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 229, in update_source
2017-06-21 10:03:37,314 ! raise RuntimeError("Could not update, \"git pull\" failed with returncode {}".format(returncode))
2017-06-21 10:03:37,315 ! RuntimeError: Could not update, "git pull" failed with returncode 1
`

jarrodasmith, 2017-10-26:
I've now reinstalled from scratch without first manually installing the libyaml-dev packages.  Here is the error I saw that led me to submit this "bug":

2017-10-26 02:34:45,459 > Searching for PyYAML>=3.10,<3.11
2017-10-26 02:34:45,461 > Reading https://pypi.python.org/simple/PyYAML/
2017-10-26 02:34:45,463 > Best match: PyYAML 3.10
2017-10-26 02:34:53,247 ! build/temp.linux-armv6l-2.7/check_libyaml.c:2:18: fatal error: yaml.h: No such file or directory
2017-10-26 02:34:53,251 ! #include <yaml.h>
2017-10-26 02:34:53,255 ! ^
2017-10-26 02:34:53,260 ! compilation terminated.
2017-10-26 02:34:54,687 !

But as you say it continues on and completes the install successfully without the extension.  My apologies for raising a red herring.

Jarrod

Tiefschneegiftler, 2017-10-26:
Hi foosel!
So the link to the regular log entries still is the same as above: https://gist.github.com/Tiefschneegiftler/ecdafdbb99685551342063984f8b2f70

However there is that dialog box that pops up:
https://gist.github.com/Tiefschneegiftler/f0f59af15d147920e51cb9bcae8e890a

And here is the "update log":
https://gist.github.com/Tiefschneegiftler/3fdbaa955f95795a34af658fc5215bc1

Anything else I may supply you with?

normanclanger, 2017-10-27:
Hi - I'm running Octoprint on a RaspberryPi, attempting to upgrade from 1.3.4 to 1.3.5 because of a prompt that pops up in the GUI.  To date, updates within the GUI have always been successful (possibly there was a 1.2.7 manual step required).

Today I get a repeated failure - the log entry is as below.  Any clues please?

thanks

Tim

Updating, please wait.
'build/scripts-2.7' does not exist -- can't clean it
Python executable: '/home/pi/OctoPrint/venv/bin/python'
>>> Running: git diff --shortstat
> git diff --shortstat
>>> Running: git fetch
> git fetch
>>> Running: git pull
> git pull
Already up-to-date.
>>> Running: python setup.py clean
> /home/pi/OctoPrint/venv/bin/python setup.py clean
running clean
removing 'build/lib.linux-armv7l-2.7' (and everything under it)
removing 'build/bdist.linux-armv7l' (and everything under it)
removing 'build'
recursively removing *.pyc from 'src'
removing 'me/pi/OctoPrint/src/octoprint_setuptools/__init__.pyc'
>>> Running: python setup.py install
> /home/pi/OctoPrint/venv/bin/python setup.py install
running install
running bdist_egg
running egg_info
writing requirements to src/OctoPrint.egg-info/requires.txt
writing src/OctoPrint.egg-info/PKG-INFO
writing top-level names to src/OctoPrint.egg-info/top_level.txt
writing dependency_links to src/OctoPrint.egg-info/dependency_links.txt
writing entry points to src/OctoPrint.egg-info/entry_points.txt
reading manifest file 'src/OctoPrint.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'src/OctoPrint.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
creating build
creating build/lib.linux-armv7l-2.7
creating build/lib.linux-armv7l-2.7/octoprint
creating build/lib.linux-armv7l-2.7/octoprint/templates
creating build/lib.linux-armv7l-2.7/octoprint/templates/_data
creating build/lib.linux-armv7l-2.7/octoprint_client
copying src/octoprint_client/__init__.py -> build/lib.linux-armv7l-2.7/octoprint_client
copying src/octoprint/events.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/daemon.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/settings.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/__init__.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/_version.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/__main__.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/users.py -> build/lib.linux-armv7l-2.7/octoprint
copying src/octoprint/timelapse.py -> build/lib.linux-armv7l-2.7/octoprint
creating build/lib.linux-armv7l-2.7/octoprint_setuptools
copying src/octoprint_setuptools/__init__.py -> build/lib.linux-armv7l-2.7/octoprint_setuptools
creating build/lib.linux-armv7l-2.7/octoprint/printer
copying src/octoprint/printer/profile.py -> build/lib.linux-armv7l-2.7/octoprint/printer
copying src/octoprint/printer/standard.py -> build/lib.linux-armv7l-2.7/octoprint/printer
copying src/octoprint/printer/__init__.py -> build/lib.linux-armv7l-2.7/octoprint/printer
copying src/octoprint/printer/estimation.py -> build/lib.linux-armv7l-2.7/octoprint/printer
creating build/lib.linux-armv7l-2.7/octoprint/plugins
copying src/octoprint/plugins/__init__.py -> build/lib.linux-armv7l-2.7/octoprint/plugins
creating build/lib.linux-armv7l-2.7/octoprint/cli
copying src/octoprint/cli/dev.py -> build/lib.linux-armv7l-2.7/octoprint/cli
copying src/octoprint/cli/server.py -> build/lib.linux-armv7l-2.7/octoprint/cli
copying src/octoprint/cli/config.py -> build/lib.linux-armv7l-2.7/octoprint/cli
[... 1122 lines cut ...]
Traceback (most recent call last):
File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>
main()
File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 310, in main
install_source(python_executable, folder, user=args.user, sudo=args.sudo)
File "/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 254, in install_source
raise RuntimeError("Could not update, \"python setup.py install\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "python setup.py install" failed with returncode 1
copying src/OctoPrint.egg-info/requires.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
copying src/OctoPrint.egg-info/top_level.txt -> build/bdist.linux-armv7l/egg/EGG-INFO
creating 'dist/OctoPrint-1.3.5-py2.7.egg' and adding 'build/bdist.linux-armv7l/egg' to it
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing OctoPrint-1.3.5-py2.7.egg
removing '/home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg' (and everything under it)
creating /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg
Extracting OctoPrint-1.3.5-py2.7.egg to /home/pi/OctoPrint/venv/lib/python2.7/site-packages
OctoPrint 1.3.5 is already the active version in easy-install.pth
Installing octoprint script to /home/pi/OctoPrint/venv/bin
Installed /home/pi/OctoPrint/venv/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg
Processing dependencies for OctoPrint==1.3.5
Searching for futures>=3.1.1,<3.2
Reading http://pypi.python.org/simple/futures/
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
The update did not finish successfully. Please consult the log for details

foosel, 2017-10-27:
@Tiefschneegiftler 

```
2017-10-24 08:29:49,957 > 191 files changed, 93857 deletions(-)
```

This indicates that your checkout is seriously compromised. It's basically showing a diff of the complete code base there, which looks like the checkout on your Pi is pretty much empty.

```
error: unable to resolve reference refs/remotes/origin/master: Invalid argument
```

And this screams corrupted git folder. Did you by any chance not shutdown your Pi correctly in the past? Because that certainly looks like some serious data corruption on your SD card. I'd suggest to make a list of your plugins, backup your `~/.octoprint` folder and then start with a clean install, because if this much is broken in your filesystem just in that folder, it's unclear what else might be.

@normanclanger 

```
Reading http://pypi.python.org/simple/futures/
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
```

Your's looks like network problems while it's trying to fetch some dependencies from the Python Package Index. Any network issues on your location recently?

---

In any case, both of you are running into issues outside of OctoPrint's control. I'm happy to help you trying to debug them further as time permits, but I think at this point the official support channels in form of the [Mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741) would actually be a better place than a bug tracker.

And the original issue reported by @jarrodasmith (which those two above are completely unrelated two) has turned out to be a red herring as well.

Therefore I'm closing this.

closed 2017-10-27 by foosel

normanclanger, 2017-10-27:
Thanks - I burnt a new sd card using octopi, and the upgrade then happened successfully.

Tiefschneegiftler, 2017-10-30:
Thx @foosel Works like a charm now. :D