OctoPrint/OctoPrint #2159: Upload Failed: 413 Request Entity Too Large

ctag, 2017-10-19:
#### What were you doing?

Uploading a larger-than-normal gcode file (2.2MB) results in a failed upload and the error "Upload Failed | 413 Request Entity Too Large"

On previous Octoprint versions I've been able to upload files this large and larger. I'm prefacing that I am uncertain whether this error is caused by upgrading or by misconfiguration.

#### What did you expect to happen?

Successful file upload.

#### What happened instead?

File upload fails both from UI and API call from Slic3r.

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5.post0.dev0+g77753ca0 (master branch)

#### Operating System running OctoPrint

Archlinux-arm, Raspberry Pi 3

#### Printer model & used firmware incl. version

Printrbot Simple Metal

#### Browser and Version of Browser, Operating System running Browser

Chromium 61, Archlinux

#### Link to octoprint.log

https://gist.github.com/ctag/ca53be0c50c1eef13678fde6383c46b4

#### Link to contents of terminal tab or serial.log

Appears to be N/A

#### Link to contents of Javascript console in the browser

https://gist.github.com/ctag/d24519e93c4a254f063028ec09748088

#### Screenshot(s)/video(s) showing the problem:

Appears to be N/A

I have read the FAQ.

Thanks.

foosel, 2017-10-20:
Hm... I just uploaded a 5MB GCODE file without any issues to a 1.3.5 instance. The upload endpoint by default should allow uploads of up to 1GB (unless you've changed your config, check if `server.uploads.maxSize` is set in config.yaml and if so to a lower value than 1073741824). There were no changes in the upload mechanism either between 1.3.4 and 1.3.5. 

Two things I noticed here:

> OctoPrint 1.3.5.post0.dev0+g77753ca0 (master branch)

That's not a clean 1.3.5 install, something in your sources is modified compared to stock 1.3.5. I'd investigate what that is to make sure that's not the reason here.

Your logs also indicate that you are connecting via an external URL. To rule out anything in between interfering (e.g. a reverse proxy which limits payload sizes), try a direct connection (localhost or at least direct LAN connection).

foosel, 2017-10-20:
And a third thing I just noticed: no log line about your failing upload attempts, which makes it sound like those don't even reach OctoPrint, making the "something in between going wrong" guess more likely.

ctag, 2017-10-20:
@foosel,

Thank you. I apologize for having you troubleshoot 3rd party software for me.

The issue was being caused by my Nginx proxy, which I had forgotten was a part of the mix. Adding `client_max_body_size 1G;` to the Nginx configuration file allows uploads as expected.

closed 2017-10-20 by ctag