OctoPrint/OctoPrint #2161: [Request] Live Z Offset control

Leesam, 2017-10-20:
I recently used matter control to run a print while i was reinstalling octoprint. It had a really useful feature which would be awesome to see in octoprint.

While Matter Control is running a print from gcode the controls still have 0.1 Z enabled. While printing you can raise or lower z offset in very small increments. This makes tuning up an auto level printer very easy.

foosel, 2017-10-20:
What GCODE commands are used for that? In which firmwares/firmware versions are they available?

Leesam, 2017-10-20:
I don't believe they are issuing the usual Z probe offset of M851. I believe what it is doing is adding or subtracting selected z offset values to each G1 it sends to the printer.

galvesribeiro, 2019-10-04:
Hey folks, did anyone found a workaround for it? I have the live Z Axis on the Creality CR10S Max display, but if I'm using Octaprint, the display don't show the current printing work so I can't adjust it on the printer itself... Any idea how to do that on Octoprint while the work is going?

Thank you!

jneilliii, 2019-10-05:
@galvesribeiro, you can go to the terminal tab and enter the commands @jlipavsky mentioned above.

http://marlinfw.org/docs/gcode/M290.html

galvesribeiro, 2019-10-05:
@jneilliii thanks for the reply.

However, it is not practical to go to a terminal (even in a browser) and add gcode for each adjustment you make todo on-the-fly. 

The touch panel of the creality has this functionality so it is quite hard to believe that the only way here is by using manual gcode on the terminal.

I guess the only option we have is to create a plugin “just” to do this basic thing that should be common on every print job...

jneilliii, 2019-10-05:
Yeah, a plugin would make sense to add buttons to the control tab, but I don't necessarily think it should be core to the product.  The reason why is because not all firmware has babystepping capability.

jneilliii, 2019-10-05:
Here's a quick fix.  Add this to your config.yaml.
```
controls:
  - name: Baby Stepping
    layout: vertical
    children:
    - name: Up
      command: M290 Z%(distance)s
      input:
      - default: .25
        name: Distance
        parameter: distance
    - name: Down
      command: M290 Z-.25
      input:
      - default: .25
        name: Distance
        parameter: distance
```

LazeMSS, 2021-03-31:
Can be fixed by either @jneilliii  solution or using plugins:
https://plugins.octoprint.org/plugins/terminalcommandsextended/

closed 2021-03-31 by LazeMSS