OctoPrint/OctoPrint #2164: 1.3.5 GCode Viewer Zoom on model doesn't do anything

pixelk0, 2017-10-21:
#### What were you doing?

1. I was printing my first model after updating to 1.3.5
2. clicked on GCode Viewer
3. checked Zoom in on model

#### What did you expect to happen?

It should Zoom on the model to show the model and a litle bit of the surrounding.
On a model smaller than the print bed, the zoom may be substantial.

#### What happened instead?

The view zoom didn't change at all.

#### Did the same happen when running OctoPrint in safe mode?

I didn't try safe mode

#### Branch & Commit or Version of OctoPrint

1.3.5 release

#### Operating System running OctoPrint

OctoPi 0.13.0

#### Printer model & used firmware incl. version

Anet A8

#### Browser and Version of Browser, Operating System running Browser

Chrome Version 62.0.3202.62

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

https://pastebin.com/MtZkc9Uf

#### Link to contents of Javascript console in the browser

https://pastebin.com/vLYWe8e0

#### Screenshot(s)/video(s) showing the problem:

Before
http://stuff.knackes.com/dld/201710/ScreenCap-20171021-155709_0E9585D6.jpg
After
http://stuff.knackes.com/dld/201710/ScreenCap-20171021-155733_FC3C195D.jpg

I have read the FAQ.

foosel, 2017-10-23:
Please provide a file with which to reproduce this issue (it works fine here).

Also please test in safe mode.

pixelk0, 2017-10-23:
Thank you for your answer.
Now for another part it works. (also tested with a third and the original)
I really don't know what I did differently.
I think we can close the issue

closed 2017-10-23 by pixelk0