OctoPrint/OctoPrint #2165: Timelapse configuration

autonumous, 2017-10-21:
#### What were you doing?
Just upgraded an existing installation from 1.3.4 to 1.3.5,
Checking tabs/configs to see whats changed, notice the timelapse configuration showed “default install”  settings, i.e,  
Time Lapse Mode: Off 
etc
And previous “Finished Timelapses” list empty. 

if i selected appropriate mode from dropdown, then hit “save Changes” the “Finished Timelapes” is populated with the list of previous recordings

even when selecting “Save as Default” option and restarting the same behaviour is observed.  

However, on the left “state” it does display appropriate status,
e.g.  TimeLapse: On Z Change

confirmed that config.yaml is appropriately updated.

Therefore appears to be a “view” issue.

I have tried with my existing build.  Additionally on a freshly installed octopi with minimal configuration.   i.e. a default octopi 0.14.0 using octoprint 1.3.4 later upgrading to 1.3.5.  Only after upgrade did the issue occur.
 
#### What did you expect to happen?
I expect to be able see the currently saved configuration, i.e. the mode and all settings plus existing completed timelapse recordings

#### What happened instead?
on startup, the timeless configuration section is not reflecting the settings in the config.yaml file.  It’s displaying the default ‘vanilla’ settings.
No previously recorded videos are displayed, simply a empty list.

#### Did the same happen when running OctoPrint in safe mode?
Yes

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.5 (master branch)

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version

e3D BigBox
Marlin 1.1.6 

#### Browser and Version of Browser, Operating System running Browser
osx 10.11.6
safari 11.0
chrome 61.0

#### Link to octoprint.log

[On gist.github.com or pastebin.com. ALWAYS INCLUDE and never truncate.
The Contribution Guidelines tell you where to find that.]

#### Link to contents of terminal tab or serial.log

[On gist.github.com or pastebin.com. If applicable, always include if unsure or
reporting communication issues. Never truncate.

serial.log is usually not written due to performance reasons and must be
enabled explicitly. Provide at the very least the FULL contents of your
terminal tab at the time of the bug occurrence, even if you do not have
a serial.log (which the Contribution Guidelines tell you where to find).]

#### Link to contents of Javascript console in the browser

[On gist.github.com or pastebin.com or alternatively a screenshot. If applicable -
always include if unsure or reporting UI issues.

The Contribution Guidelines tell you where to find that.]

#### Screenshot(s)/video(s) showing the problem:

[If applicable. Always include if unsure or reporting UI issues.]

I have read the FAQ.

ntoff, 2017-10-21:
Not seeing that behaviour here, please include the requested logs (although the serial log probably isn't totally necessary in this case) and browser console output.

autonumous, 2017-10-21:
Just being playing around with my set up.  With a  completely clean install it’s working fine.

However, for my setup,  I mount an external USB drive formatted as exfat on /home/pi/.octoprint/timelapse.  (using exfat-fuse)

The USB drive is mounted ok, i can read/write to that filesystem from the cli.  

Whilst “timelapse” is using the external filesystem, I have the issues as originally described.   
If i unmount the external Filesystem, and use the local/root filesystem, it behaves as expected. 

This setup worked 100% prior to upgrading octoprint. 

I’ll get the logs

autonumous, 2017-10-21:
Ok, I've resolved it!  corrupt file system on my external drive.  

tailing the octoprint.log while accessing GUI, spotted..
```
OSError: [Errno 5] Input/output error: '/home/pi/.octoprint/timelapse/tmp'
```
further confirmed by me being unable to read the contents of the tmp directory within the external USB drive.  i.e. once mounted /home/pi/.octoprint/timelapse/tmp 
```
pi@octopi:~/.octoprint/timelapse/old_tmp $ ls -ltr
ls: reading directory .: Input/output error
total 0
pi@octopi:~/.octoprint/timelapse/old_tmp $ 
```
fsck reports filesystem errors, but don’t think it can repair exfat

Funny it never caused an issue on 1.3.4.  Guessing 1.3.5 does some clean up and is unable to access that filesystem.   I should have checked the logs before hand - sorry!

Following a rename and a restart all seems well.

Will close out.

closed 2017-10-21 by autonumous