OctoPrint/OctoPrint #2166: Drag and Drop overlay doesn't go away properly

thrasr, 2017-10-22:
Dragging a file from an explorer window that partially covers OctoPrint will enable the drag/drop overlay.  Normally, dragging the file outside OctoPrint will disable the drag/drop overlay.  However if you bring the file back over the explorer window then the drag/drop overlay will remain.  Even after letting go of the file, the overlay remains and OctoPrint needs to be refreshed to be used again.

This only happens with Firefox.  OctoPrint version is the latest at 1.3.5.

[Here is an example of how to reproduce.](https://i.imgur.com/pTUre0P.png)  Drag file from window over OctoPrint and back.

foosel, 2017-10-23:
Huh, looks like that is a [browser bug](https://bugzilla.mozilla.org/show_bug.cgi?id=656164). I've just pushed a workaround to `maintenance` (and will soon also merge that to `devel`). Will be pushed as part of the 1.3.6 release.

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel