OctoPrint/OctoPrint #2168: Updated Plugin and suddently cannot login

saboaua, 2017-10-22:
Received a notification to update OctoPrint-TPLinkSmartplug and did the update, when Octoprint was reloaded asked for user and password. Enter the information and doesn't log in or show a error screen if password is wrong.

How could solve this issue that now cannot log in anymore?

jneilliii, 2017-10-22:
This would be an issue with my plugin, not the core octoprint.  It's related to the plugin settings not getting cleared out as anticipated with the on_settings_migrate hook.  There is an open ticket on my repository [here](https://github.com/jneilliii/OctoPrint-TPLinkSmartplug/issues/19). My initial thought is just restarting the service should clear things up.  Let me know.

foosel, 2017-10-23:
Not a problem with OctoPrint, but a third party plugin. Closing.

closed 2017-10-23 by foosel