OctoPrint/OctoPrint #2170: OctoPrint web GUI Login Issue

redsoxfn820, 2017-10-23:
**What were you doing?**
Using the login on the octoprint main web GUI page

**What did you expect to happen?**
The system to log me in and allow me to control my printer as usual.

**What happened instead?**
System thinks, then loads the octoprint GUI again and the url shows a login but the GUI will not enable controls and the tab still shows "Login" instead of "Logout" (See screenshot)

**Did the same happen when running OctoPrint in safe mode?**
Yes

**Branch & Commit or Version of OctoPrint**
OctoPrint 1.3.5 (master branch)

**Operating System running OctoPrint**
OctoP Jessie.

**Printer model & used firmware incl. version**
Prusa I3 w/ Repetier

**Browser and Version of Browser, Operating System running Browser**
Google Chrome Version 61.0.3163.100
Windows 10 Creators Update

**Screenshot(s)/video(s) showing the problem:**
[image]

I have read the FAQ.

ntoff, 2017-10-23:
try to SSH in and `sudo service octoprint restart` and then ctrl + F5 in your browser 

also before anything can be looked into, you'll need to provide those logs that are requested, along with the javascript terminal in your browser.

foosel, 2017-10-23:
Please provide _all_ requested information. Without that information this ticket cannot be looked into and will be closed. Thanks.

schnubbie, 2017-10-23:
I had the same problem. I checked my logs and find out that the problem be caused by the TPSmartlink-Plugin. I unistalled it by ssh and now all is working fine again.

jneilliii, 2017-10-23:
I could use some help on this with my plugin.  I thought I had done the on_settings_migrate function correctly initially by setting a calue to null, but that didn't work, so I modified the latest branch, version 0.7.1 to set the setting to the equivalent defaults using self._settings.set(['arrSmartplugs'], self.get_settings_defaults().arrSmartplugs).  Please help @foosel.

redsoxfn820, 2017-10-24:
@schnubbie how did you uninstall it via ssh? what was the command?

iceblu008, 2017-10-24:
> @schnubbie how did you uninstall it via ssh? what was the command?

https://github.com/foosel/OctoPrint/wiki/FAQ#i-installed-a-broken-plugin-how-do-i-uninstall-it-again

redsoxfn820, 2017-10-24:
Thanks for the help @iceblu008 and @schnubbie uninstalling TPLinkSmartplug has solved the issue.

foosel, 2017-10-24:
@jneilliii I just pinged you on your own ticket on this.

foosel, 2017-10-24:
@redsoxfn820 if uninstalling that plugin indeed solved your issue, you posted some misleading information:

> **Did the same happen when running OctoPrint in safe mode?**
> Yes

It's ok to write that you couldn't try safe mode due to this issue, but if you'd really started OctoPrint in safe mode you'd not been able to reproduce this issue there - which is very very important information to debug stuff like that. So please, if you get asked to try something in safe mode and you can't, say you can't, not that you tried and the issue was still showing there too. 

In any case, this is a duplicate of #2168 and actually an issue with a third party plugin, not OctoPrint - see jneilliii/OctoPrint-TPLinkSmartplug#19. Since more people seem to run into this and 0.7.1 of the plugin also doesn't seem to solve this problem though, I'll leave this open for a bit so that people actually see it instead of opening new tickets about this all the time.

BillyBlaze, 2017-10-24:
@foosel , perhaps it's a good idea if we change event on the `<form>` of the post to something like this?: 

````
<form id="loginForm" data-bind="event: {'submit': loginState.prepareLogin }" onsubmit="return false;">
````

This to make sure we can prevent posting the page with GET even if knockout fails

foosel, 2017-10-24:
@BillyBlaze see 0f8f4579d6bf5afdaaa1217975fd02bbb2da49ee :D Already prepped for 1.3.6.

BillyBlaze, 2017-10-24:
You're awesome! 👍

redsoxfn820, 2017-10-24:
@foosel yes that was my fault, it was a copy paste error.  Sorry for the incorrect posting location as well!

Sounds good, Hopefully this will help others who installed the TPLinkSmartplug plugin

jneilliii, 2017-10-24:
Sorry for all the headaches guys...that's what I get for a being a hack programmer...lol.

kantlivelong, 2017-11-30:
Is this resolved? Can this ticket be closed?

foosel, 2017-11-30:
Hey, good point... it can!

closed 2017-11-30 by foosel