OctoPrint/OctoPrint #2171: Login Function of web page unresponsive

iceblu008, 2017-10-24:
I was updating from 1.3.4 master branch to 1.3.5 master branch thru the automatic updater notification that comes up on the webpage

I expected a smooth transition to a new version of octoprint

I updated and now my octoprint is rendered useless and is not functional.

It is not possible to run Octoprint in safe mode because I can't log in.

updating to 1.3.5 master branch

OctoPI 1.3.4 to 1.3.5, Linux, Raspberry pi 3

Robo R1 plus

Google Chrome Version 61.0.3163.100 on Windows 10(Tried multiple browsers)

N/a

https://pastebin.com/EGrPaajK

N/a

I have read the FAQ.

foosel, 2017-10-24:
> It is not possible to run Octoprint in safe mode because I can't log in.

Doing it from the UI is only one of THREE options to start safe mode. See http://docs.octoprint.org/en/master/features/safemode.html

Your problem is most likely caused by the TPLinkSmartplugin plugin, see jneilliii/OctoPrint-TPLinkSmartplug#19. [Uninstall it manually through SSH]() or when running safe mode.

Closing as duplicate of #2168

foosel, 2017-10-24:
Also next time please fill out the full ticket template - with the amount of information you provided it would have been absolutely impossible to even get an inkling of an idea of what might be going wrong if I hadn't known about the issues with that specific plugin already. Every single bit of information that is asked for is asked for for a reason. Having to run after this information all the time takes up time that would be **way** better spent on fixing reported issues.

closed 2017-10-24 by foosel