OctoPrint/OctoPrint #2175: OctoPrint does not cancel printing on external reset

AtoxIO, 2017-10-25:
#### What were you doing?

I was printing a print when I noticed spaghetti, so I decided to reset it. Since I was closer to the printer than to the PC, I pressed the physical reset button. 

#### What did you expect to happen?

For OctoPrint to stop printing.

#### What happened instead?

OctoPrint detected the reset, but continued the print anyway, because the printer lost position this caused the effector to crash into the tower, nearly breaking the effector arms.

#### Did the same happen when running OctoPrint in safe mode?

Yes, also I created a special testing gcode (move in a straight line) to prevent danger to my printer.

#### Branch & Commit or Version of OctoPrint

Version 1.3.5 (master branch)

#### Operating System running OctoPrint

Octopi 0.14.0

#### Printer model & used firmware incl. version

Kossel-style Delta Printer, Repetier 0.92.9

#### Link to octoprint.log

https://gist.github.com/AtoxIO/140253dfae35cc9bf3146631c9a39e2d

#### Link to contents of terminal tab or serial.log

https://gist.github.com/AtoxIO/a35fdafc20c13e38ab609ebdd4f6a3a6

#### Link to contents of Javascript console in the browser

https://gist.github.com/AtoxIO/3805a288ad3d9d3ec5c1c9b8c0cf0492

#### Screenshot(s)/video(s) showing the problem:

Can make one if you really need me to.

I have read the FAQ.

foosel, 2017-10-26:
I was just about to look into fixing this when I realized it was already fixed in 4d27de032d4f749a6c942bfd6d8fca0207df8827 😊 That fix is already available on `maintenance` and `devel` and will be released with 1.3.6.

foosel, 2017-12-12:
Closing because 1.3.6 was just [released](https://octoprint.org/blog/2017/12/12/new-release-1.3.6/).

closed 2017-12-12 by foosel