OctoPrint/OctoPrint #2180: First connection attempt fails, disconnect/connect then succeds

keepiru, 2017-10-26:
The first attempt to connect to my printer via serial always fails.  Repeated attempts usually work.  Once a connection succeeds, disconnect/reconnect usually works.  However, if I wait until "Connection closed, closing down monitor" appears, the next connection will fail, requiring a disconnect/connect cycle to fix it.

Bug #1770 sounds similar, except I do not see the "Recv: wait" lines.

#### What were you doing?
1.  Power on OctoPi.
2.  Wait for UI to be available.
3.  Click "Connect".
4.  Wait 10 seconds.
5.  Click "Disconnect", then one second later click "Connect".
6.  GOTO 4 if it doesn't connect.

#### What did you expect to happen?
Connection in step 3.

#### What happened instead?
Connection hangs until second connection attempt in step 5.

#### Did the same happen when running OctoPrint in safe mode?
Yes, identical behavior.

#### Branch & Commit or Version of OctoPrint
OctoPrint 1.3.5 (master branch)

#### Operating System running OctoPrint
OctoPi 0.14.0

#### Printer model & used firmware incl. version
MP Select Mini v2, motion firmware v40.  M115 reports:
2017-10-26 00:19:51,844 - Recv: NAME. Malyan    VER: 4.0        MODEL: M200     HW: HH02
2017-10-26 00:19:51,848 - Recv: CLK:EXT BUILD: Aug 14 2017 16:46:00

#### Browser and Version of Browser, Operating System running Browser
Chromium 57.0.2987.98 Built on 8.7, running on Debian 8.9 (64-bit)

#### Link to octoprint.log
https://gist.github.com/keepiru/b7f67936b0989d3db3562a747b5c9b1e

#### Link to contents of terminal tab or serial.log
In this case it took two disconnect/reconnect cycles before it worked. 

Failure 1: https://gist.github.com/keepiru/eca1e82e227a1b4d44b3dd43900b67cc
Failure 2: https://gist.github.com/keepiru/c1a6f8e5b916ff4bf6522a8bb7dcf12b
Success: https://gist.github.com/keepiru/8803eff2ebb07d27f57b11454578d24d

Also note the interesting "2017-10-26 00:26:47,891 - Connection closed, closing down monitor" in the middle of the successful connection in the last log.

#### Link to contents of Javascript console in the browser
/dev/null

#### Screenshot(s)/video(s) showing the problem:
/dev/urandom

I have read the FAQ.

foosel, 2017-11-03:
This looks like your printer for some reason simply doesn't respond to the initial `M110` in a timely manner unless you do a bunch of reconnects first. Which sounds odd. 

What does the printer do when you connect, does it reset (e.g. LCD goes off, some logo shows, something like that) or does it just keep doing what ever it is doing?

What does happen if you do *not* hit Disconnect manually? 

Your final `serial.log` doesn't look like it does reset (none of the normal reset noise like `start`, firmware version and such), but this being a Malyan with its self-tinkered firmware that does its own thing all the time, that sadly doesn't say much.

foosel, 2017-11-23:
Closing due to lack of response.

closed 2017-11-23 by foosel

keepiru, 2017-11-25:
After some more testing I've found the problem is definitely not on OctoPrint's side.  The same behavior occurs using Picocom.

However, it would be nice if there was an option to work around this by automatically close and reopen the port if there's difficulty connecting.  Can we make this a feature request?

kantlivelong, 2017-11-25:
This could be accomplished by a plugin fairly easily.