OctoPrint/OctoPrint #2181: Can't connect to printer since octoprint 1.3.3

jting1981, 2017-10-26:
#### What were you doing?

Connecting to my printer with latest Octoprint version.

But downgraded to Octoprint1.3.2 works fine.

#### What did you expect to happen?

Connect successfully.

#### What happened instead?

Alaways fail connect and shown "Unhandled communication error" whatever the Baudrate is "AUTO" or "115200"

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.5 (master branch)

But Octoprint 1.3.2 (master branch) connect successfully

#### Operating System running OctoPrint

OctoPi.

#### Printer model & used firmware incl. version

InfinityX1 (Not sale in US or EU or China)
http://www.infinity3dp.com

This is the firmware file, I don't know if this help.
http://www.infinity3dp.com/upload/download/1503374510.zip

#### Browser and Version of Browser, Operating System running Browser

Google Chrome 61.0.3163.100, OS:Windows 7 64bit

#### Link to octoprint.log

https://pastebin.com/Ebt8G84b

#### Link to contents of terminal tab or serial.log

serial.log is empty.

terminal tab:
https://pastebin.com/3u6GWzM2

#### Link to contents of Javascript console in the browser

https://pastebin.com/Cc3S3tPX

#### Screenshot(s)/video(s) showing the problem:

https://imgur.com/cVmmTXt

https://imgur.com/uIa9i9d

foosel, 2017-10-26:
It looks like your printer is putting out a non-standard start line and that's confusing OctoPrint because it looks like the response to an `M115` command, but doesn't fully match that either. That causes an error in OctoPrint's parsing routine for that specific command.

While that's easy to fix, your printer vendor should seriously check their firmware's behaviour there too - it's spewing binary garbage on initial connect (`\x00`), then doesn't send a `start` but instead a misconstructed `M115` response without being prompted to do so... If I were you, I'd also get in touch with my printer/firmware vendor because if something as simple as the initial connect already has these many things that differ from established de-facto standards, chances are high you'll run into further issues some day down the road as well.

foosel, 2017-10-26:
Should be fixed by the above commit, already on `maintenance`, soon on `devel`, will be released with 1.3.6.

jting1981, 2017-10-26:
Thank you, I'll talk the printer vendor, and tell them to treat this matter seriously. This confused me for long time. With your reply, I can more confident to talk to them.

commit a1548e1c (2017-10-26): Fix issue in M115 parser

If FIRMWARE_NAME is not there, but NAME is neither, we'd run into an
exception due to the Malyan detection. Not anymore.

Fixes #2181
closed 2017-12-12 by foosel