OctoPrint/OctoPrint #2188: [Request] Adding options  fields for timelapse FFMPEG

MasterPIC, 2017-10-30:
Currently there is the ability to flip, mirror and rotate but it's limited to webcam.
/usr/bin/avconv can accept options to rotate or flip (e.g. **-vf "hflip,vflip"**).
So what about allowing the user to set some options to pass to FFMPEG bin?

Thanks
Armando

ntoff, 2017-10-30:
so are your timelapse videos coming out upside down? because I'm pretty sure if you've ticked the box to flip the webcam, ffmpeg also gets sent the flip parameter https://github.com/foosel/OctoPrint/blob/0a69dbeddb301d5a32827a3f0d561f875df24234/src/octoprint/timelapse.py#L811

MasterPIC, 2017-10-30:
The options available for webcam don't affect the shots used for timelapse.

MasterPIC, 2017-10-30:
I'm going to check again. According to source the options should affect timelapse too.

foosel, 2017-10-30:
Actually, they do and @ntoff even linked the code that is used to do that. See also https://github.com/foosel/OctoPrint/blob/0a69dbeddb301d5a32827a3f0d561f875df24234/src/octoprint/timelapse.py#L775-L789

If they don't that's a bug.

ntoff, 2017-10-30:
OctoPrint has precisely *zero* control over how the images come out of  your webcam. The webcam is handled by mjpeg streamer and the options it supports for your particular camera type (raspi cam supports more options than a usb web camera does).

MasterPIC, 2017-10-30:
OK, it seems video and shots orientations are not the same in my case: I'm using IP webcam on Android and it can flip/rotate **only** the video, not the shots. On the video there is some overlayed text.
So video is fine, shots aren't...
To fix the shots (and so the timelapse) I need dedicated options to pass to avconv...
I know it's a particular case. Anyway I see there is the ability to use different sources to take live video and timelapse shots. So maybe it's still worth adding (timelapse) dedicated options.

ntoff, 2017-10-30:
I believe there is a brainstorming thread around here somewhere about more advanced timelapse features, so yes advanced command line options would definitely be something to consider for a future version of octoprint.

Stupid question but can you just flip your phone (I assume it's a phone) so it's the other way up and then use OctoPrint's flip commands? Or does that mean mounting the phone in a weird position?

I believe there's a way to use a script to take over the rendering, but the last time I tried using that was when octoprint used the old timelapse system, but if it still works then there might be a hacky way to get what you want while you wait for more advanced timelapse features in octoprint. I'll see if I can dig it up (it's 3am here right now) and get back to you.

MasterPIC, 2017-11-08:
I can't flip the phone (its flash is not aligned to the lens so, when flipped, it would not illuminate the bed because of the small distance from the printer).

chambm, 2018-01-09:
@MasterPIC In IP webcam, `/photo.jpg` is not flipped, but `/shot.jpg` is. However, my timelapse videos are still coming out upside down even though the video looks correct in OctoPrint's control view. I have the flipping configured in IP webcam itself so I don't have OctoPrint configured to do it. Which makes it all the more strange that I'm getting flipped videos... the images in the `tmp` folder during a print are flipped.

I just downloaded fiddler to try to figure out what is going on, and it says OctoPrint is still asking for `/photoaf.jpg` (autofocused photo) instead of `/shot.jpg` like it says in the configuration screen. Maybe because I changed it during a print it didn't take effect? `Config.yaml` has the current setting `/shot.jpg` though. I'm guessing OctoPrint allowed me to change the setting, but didn't change the in-memory value during a print.  It should have applied it after the print stopped though (for the next print). If this is by design, then the timelapse configuration screen should probably be disabled during a print.

This also explains why I kept losing focus in IP webcam!

MasterPIC, 2018-01-09:
@chambm /photo.jpg is a jpeg taken from photo, shot.jpg is latest video frame. And their orientations can differ.
Since Octoprint orientation settings affect both video and timelapse frames, the user should set IP Webcam to have the same orientation on video and photo frames. Then Octoprint can eventually flip them if required...
What can't be fixed is the wrong IP Webcam watermark orientation when Octoprint flip is enabled...

chambm, 2018-01-09:
I don't see where to set flipping for photo.jpg? Although I was just looking at the web config interface. The advantage of doing the flip in IP Webcam is that the video is correct in the web interface too. I do see it's possible to specify a separate resolution for photos than video; I had them set the same so that's why I didn't notice the difference between photo.jpg and shot.jpg. The biggest reason for me to use shot.jpg is that it doesn't interfere with the focus. I have it set to manual focus because autofocus is annoying. :)

MasterPIC, 2018-01-09:
> I don't see where to set flipping for photo.jpg

In fact there is no way to change IP Webcam photo.jpg orientation, you can change only IP Webcam video orientation. This is exactly the reason why you should make their orientation consistent and, eventually, if the common orientation is wrong, fix it by Octoprint.
But Octoprint orientation fix will affect IP Webcam overlay when enabled...

closed 2022-02-17 by cp2004