OctoPrint/OctoPrint #2190: [Request] Display actual tool and bed temperature on Control page

lairdb, 2017-10-30:
It would be convenient and useful to display the tool and bed temperature on the "Control" tab.

If this were implemented as an image overlay, perhaps it could also optionally be fed to the timelapse feed.

If it were fed to the timelapse, ZHeight might also be interesting to see.

ntoff, 2017-10-30:
There's a plugin that shows them up on the navbar http://plugins.octoprint.org/plugins/navbartemp/ so they're always visible

token47, 2018-01-07:
Yes, put them in the navbar resolved this very issue to me. But then I got to the other issue that the navbar is not sticky to the top and you have to scroll to see it, frequently. There's another issue/request for making the navbar sticky to the top, let's hope some developer find it useful and make a plugin for that.

ntoff, 2018-01-07:
There's already a plugin for that too https://github.com/ntoff/OctoPrint-Fixednav

token47, 2018-01-07:
Nice!!! The odd thing is that you made it about 3 months ago (according to github) and I still couldn't find it until now (I even searched for a plugin like this again yesterday before posting this).

It does not appear on a search inside octoprint:
[image]

And also does not appear on the octoprint website:
[image]

Anyway, thanks! I'll try it.

jneilliii, 2018-01-07:
Oops, I didn't realize you had created one either @ntoff and spit one out in response to the other issue post. I think people don't find your plugins because they never get submitted to the plugin repository.

ntoff, 2018-01-07:
yeah there's a few I make and don't register either because they're specific to my needs, or aren't finished / polished enough for the repository. I thought I had registered the sticky navbar one though, but I guess not. Go ahead and register yours if you  want.

LazeMSS, 2021-03-31:
This is fixed by way of plugins:
https://plugins.octoprint.org/search/?q=temp

closed 2021-03-31 by LazeMSS