OctoPrint/OctoPrint #2191: [Request] M118 - Display a message in the host console for the virtual printer

MrWyss, 2017-10-30:
would allow us to test action: commands
http://reprap.org/wiki/G-code#M118:_Echo_message_on_host

e.g. 
Send: M118 //action:lightson
Recv: //action:lightson

foosel, 2017-10-30:
That's been possible for a while with `!!DEBUG:send //action:lightson` or in your specific case even longer with `!!DEBUG:action_custom lightson`. If someone wants to adds what boils down to what would basically be an aid for that though, a pull request against the maintenance branch for that is welcome.

Documentation on docs.octoprint.org was still missing until recently, but that has been fixed on maintenance, see http://docs.octoprint.org/en/maintenance/development/virtual_printer.html#debug-commands

cp2004, 2022-06-19:
## Implementation help
*This issue may be suitable for a new contributor to pick up*

- A new gcode command could be added, like this one for M117 https://github.com/OctoPrint/OctoPrint/blob/53b9b6185781c07e8c4744a6e28462e96448f249/src/octoprint/plugins/virtual_printer/virtual.py#L696-L702
- Use the `self._send` function to respond with the correct message

foosel, 2022-10-26:
Implemented by @max246 in #4567, to be part of 1.9.0. Thanks!

foosel, 2023-05-23:
1.9.0 has been released.