OctoPrint/OctoPrint #2193: Upgrade 1.3.4>1.3.5 fail : Couldn't find index page for 'futures'

AndKe, 2017-10-31:
What were you doing?
-I have tried to update octoprint via gui from 1.3.4 to 1.3.5

-What did you expect to happen?
that the update completes

-What happened instead?
the update failed

-Did the same happen when running OctoPrint in safe mode?
yes

-Branch & Commit or Version of OctoPrint
OctoPrint 1.3.4 (master branch) 

-Operating System running OctoPrint
debian 8 on Raspi 2
Log output:

Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
Couldn't find index page for 'futures' (maybe misspelled?)
No local packages or download links found for futures>=3.1.1,<3.2
error: Could not find suitable distribution for Requirement.parse('futures>=3.1.1,<3.2')
Traceback (most recent call last):
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 313, in <module>
main()
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 310, in main
install_source(python_executable, folder, user=args.user, sudo=args.sudo)
File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate/scripts/update-octoprint.py", line 254, in install_source
raise RuntimeError("Could not update, \"python setup.py install\" failed with returncode {}".format(returncode))
RuntimeError: Could not update, "python setup.py install" failed with returncode 1
The update did not finish successfully. Please consult the log for details.

ntoff, 2017-10-31:
There's a bunch of discussion on it over here https://github.com/guysoft/OctoPi/issues/427 too.

This seems to be a weird issue as people there are reporting it happens even on a fresh install, which only seems to happen for some people and not others. @foosel has done extensive testing on various upgrade paths and can't reproduce this, and I've tested a fresh install of octopi 0.14 and the upgrade goes fine for me also.

* How old is your octoprint installation?
* What's your network topology like? Are you behind any sort of proxy? Do you run any sort of proxy on your network for ad blocking or parental filters (things like caching proxies or pihole)? Does your ISP require you to configure a proxy server?
* Do you know if your ISP blocks or filters certain content? (maybe list who your ISP is?)
* Does the update actually "break" your entire pi as the people in that other discussion are stating, or does it merely break octoprint itself? Are you still able to log in via SSH after OctoPrint goes down? Or does the pi itself actually refuse to boot afterwards? I'm assuming since you have the log then your pi remains operational and you are able to SSH into it?

AndKe, 2017-10-31:
-My octopi is at about one year old.
-nothing funny about the network, no proxy, no firewall/filtering/censorship.
-I am not aware of my ISP playing big brother.
-I can ssh into it just fine, it boots, but does not serve it's web-interface anymore.

if there is a way to retry the upgrade, or in any way recover without loosing the config, I would like to try :)

ntoff, 2017-11-01:
Your config files are all saved in `/home/pi/.octoprint` so as long as you keep that directory safe, your config files should remain intact. Note the period or full stop on the front of the octoprint directory, it's a hidden directory in linux (in case you're browsing from a GUI), and is separate from `/home/pi/OctoPrint` where octoprint itself resides.

You can try the manual update outlined here https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-update-the-octoprint-installation-on-my-octopi-image but as the issue seems to be that SSL is failing, I doubt the manual install will help you either (doesn't seem to work in that other thread).

You could try to roll back to 1.3.4 https://github.com/foosel/OctoPrint/wiki/FAQ#how-can-i-revert-to-an-older-version-of-the-octoprint-installation-on-my-octopi-image and see if that works, if not then you might have to copy your octoprint config files to a safe spot (usb thumbdrive maybe) and re-flash the SD card with octopi 0.14 and then ignore the 1.3.5 update until a possible solution can be found.

Please be patient while the issue is being looked into, since @foosel can't reproduce the error, and it doesn't happen to everyone (there were no reports from any of the 1.3.5 release candidate testers, and the release candidates were out for a month? or two? before 1.3.5's final release) it might take a little while to figure out why pip / ssl is suddenly broken, and why it's only affecting some people and not others.

foosel, 2017-11-01:
> -Operating System running OctoPrint
> debian 8 on Raspi 2

OctoPi or custom install? Asking because it looks you have the `/home/pi/oprint` virtual environment present, which sounds more like OctoPi. If so, what version please? `cat /etc/octopi_version`

Also, what version of pip is this? `~/oprint/bin/pip --version`

Finally, does it get fixed by updating pip? `~/oprint/bin/pip install --index-url https://pypi.python.org/simple --upgrade pip`

AndKe, 2017-11-01:
@ntoff thanks for the suggestions, I have not tried anything yet, so I'll answer @foosel first:
Yes, it's an all-in-one OctoPi. Installed once, upgraded along the way, working fine till now.
/etc/octopi_version is 0.12.0

pip 1.1 from /home/pi/oprint/lib/python2.7/site-packages/pip-1.1-py2.7.egg (python 2.7)
upgraded and have now:
pip 9.0.1 from /home/pi/oprint/local/lib/python2.7/site-packages (python 2.7)

How can I retry the 1.3.5 upgrade now - to see if it succeeds ?

foosel, 2017-11-01:
Try this, that's basically what the update script does:

```
cd ~/OctoPrint
~/oprint/bin/python setup.py install
sudo service octoprint restart
```

AndKe, 2017-11-01:
Install still fail at the "futures" :+1: 
`creating /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg
Extracting OctoPrint-1.3.5-py2.7.egg to /home/pi/oprint/lib/python2.7/site-packages
OctoPrint 1.3.5 is already the active version in easy-install.pth
Installing octoprint script to /home/pi/oprint/bin

Installed /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg
Processing dependencies for OctoPrint==1.3.5
Searching for futures>=3.1.1,<3.2
Reading http://pypi.python.org/simple/futures/
Couldn't find index page for 'futures' (maybe misspelled?)
Scanning index of all packages (this may take a while)
Reading http://pypi.python.org/simple/
No local packages or download links found for futures>=3.1.1,<3.2
error: Could not find suitable distribution for Requirement.parse('futures>=3.1.1,<3.2')`

foosel, 2017-11-01:
But `~/oprint/bin/pip --version` still reports it's at 9.0.1? O_O

AndKe, 2017-11-01:
yes, pip 9.0.1 from /home/pi/oprint/local/lib/python2.7/site-packages (python 2.7)

buzztiaan, 2017-11-01:
> Reading http://pypi.python.org/simple/futures/
> Couldn't find index page for 'futures' (maybe misspelled?)

huh? that totally give a index page here on my browser
what happens on your installation when you open that page in a browser? maybe install links or w3m or just try with wget?  does that pi even have internet access?

foosel, 2017-11-01:
@buzztiaan It's an issue with older `pip` versions. That it works in your browser is because it redirects it to https. Doesn't work for that old pip version though.

@AndKe It appears that for some reason the old pip version stays in the virtual environment and `python setup.py install` uses that, whereas `pip --version` users the new one. Not very helpful behaviour, but let's try to get around it. Does

```
cd ~/OctoPrint
pip install .
sudo service octoprint restart
```

work? I just flashed an OctoPi 0.12 and here at least that gets things going. It's not the definitive solution however, since `python setup.py install` also needs to be able to function correctly.

AndKe, 2017-11-01:
```
pi@octopi ~/OctoPrint $ pip install .
Unpacking /home/pi/OctoPrint
  Running setup.py egg_info for package from file:///home/pi/OctoPrint
    
Downloading/unpacking flask>=0.9,<0.11 (from OctoPrint==1.3.5)
  Cannot fetch index base URL http://pypi.python.org/simple/
  Could not find any downloads that satisfy the requirement flask>=0.9,<0.11 (from OctoPrint==1.3.5)
No distributions at all found for flask>=0.9,<0.11 (from OctoPrint==1.3.5)
Storing complete log in /home/pi/.pip/pip.log

```

foosel, 2017-11-01:
Sorry, that should have been `~/oprint/bin/pip install .`

In any case, even better:

```
printf -- "[easy_install]\nindex-url = https://pypi.python.org/simple\n" > ~/.pydistutils.cfg
```

That also makes `~/oprint/bin/python setup.py install` work again.

AndKe, 2017-11-01:
@foosel the printf did it, it works fine as usual now - Thank you all

foosel, 2017-11-01:
I've pushed out an "important announcement" post on this and added an FAQ entry, that should hopefully reach those that are still running 0.12 with an unupgraded pip.

Leaving this open for a bit longer so it stays visible (Github's search by default only searches through open tickets).

RomainBouchaud, 2017-11-16:
Thank you so much @foosel , you saved my life!

seansbox, 2017-11-28:
I was having the same issue. Thanks @foosel!

closed 2018-12-10 by foosel