OctoPrint/OctoPrint #2196: Please make API key use "readonly" instead of "disabled" so it is selectable / copyable

aplocher, 2017-11-01:
Currently the API key is in a textbox marked as "disabled".  This makes it impossible to select and copy in Firefox (and probably every other browser).  The only way to enter it elsewhere is to type each character or use the QR Scan feature. 

By marking the input field as Readonly instead of Disabled, it will remain selectable.

Alternatively, a separate javascript link/button which offers to "copy" would work, too.

ntoff, 2017-11-02:
https://github.com/foosel/OctoPrint/issues/2056 API key will be getting its very own shiny "copy to clipboard" button in a future release of OctoPrint.

foosel, 2017-11-02:
That copying doesn't work due to the disabled field is indeed a Firefox-only problem as far as I am aware, see also #2056. As mentioned by @ntoff, a copy button has already been added and will be rolled out with 1.3.6.

I'm closing this since it's pretty much a duplicate of the aforementioned ticket, comments are still possible though.

closed 2017-11-02 by foosel