OctoPrint/OctoPrint #2201: "Connect" button generates error in logs?

akseanor, 2017-11-02:
#### What were you doing?
Attempting to connect to a Printrbot Play.

#### What did you expect to happen?
Octoprint would connect to the printer and allow control.

#### What happened instead?
The following error was generated in the log each time I pressed the "Connect" button:

2017-11-02 21:28:02,797 - octoprint.server.util.sockjs - INFO - New connection from client: 10.0.3.72
2017-11-02 21:28:04,346 - octoprint - ERROR - Exception on /api/printer/command/custom [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/server/api/printer.py", line 377, in getCustomControls
    customControls = settings().get(["controls"])
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/settings.py", line 1282, in get
    return self._get_value(path, **new_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/settings.py", line 1245, in _get_value
    value = preprocessor(value)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.3.5-py2.7.egg/octoprint/settings.py", line 745, in _process_custom_controls
    return map(process_control, controls)
TypeError: argument 2 to map() must support iteration
2017-11-02 21:28:04,379 - tornado.access - ERROR - 500 GET /api/printer/command/custom (127.0.0.1) 56.49ms

#### Did the same happen when running OctoPrint in safe mode?
The printer still did not connect, however no error was generated in the log.

#### Branch & Commit or Version of OctoPrint
Master branch 1.3.5

#### Operating System running OctoPrint

Latest OctoPi

#### Printer model & used firmware incl. version

Printrbot Play

#### Browser and Version of Browser, Operating System running Browser

Windows 10/Chrome

#### Link to octoprint.log
https://pastebin.com/HNMrEP5R

I have read the FAQ.

foosel, 2017-11-03:
If no error was generated in safe mode then the error originates in one of your installed third party plugins, not within OctoPrint.

Considering that your log indicates a call to a custom control instead of the actual connect API (`/api/printer/command/custom` instead of `/api/connection`) it sounds like some plugin you have installed might be rewriting the connect button. I have no idea which one that might be though.

The error itself is related to the set of custom controls not being an iterable list, which usually it should be (even if an empty one), so that might be caused by an invalid entry in your `config.yaml`, however, that doesn't explain why on pressing the connect button when not in safe mode even attempts to execute a custom control instead of the built-in connect method.

I'd suggest to disable one plugin after the other and test in between and try to figure out which specific plugin's presence is triggering this error to begin with, that might allow further diagnosis.

> **Did the same happen when running OctoPrint in safe mode?**
>
> The printer still did not connect [...]

Please provide the contents of the terminal tab when attempting to connect in safe mode. Alternatively, provide a `serial.log` from the same. You'll need to enable `serial.log` in Settings > Serial Connection first.

And in general, please always provide _all_ requested information. Without that information tickets cannot be looked into and will be closed. Thanks.

akseanor, 2017-11-03:
Apologies for not specifying, but nothing was being written to serial.log or the Terminal tab.

I actually assumed that one of the plugins might be causing the error, so I uninstalled them all prior to creating the issue, and still got the same error.

This morning I retried a few times under different conditions, and noticed that I was getting the error even under safe mode. Log starting from this morning's runs is here: https://pastebin.com/36HrnMur

The last few runs were all under Safe Mode, so it could be possible that my installation somehow got corrupted (possibly by a plugin, maybe?).

I'll look around and see if I can find a way to overwrite my install clean from the repo (not sure where all the files are on OctoPi) and report back if that fixes the issue.

akseanor, 2017-11-03:
Found some documentation and ran `git reset --hard HEAD` in the source directory, followed by the `clean` and `install` routines of setup.py, which if my thinking is right should have reverted everything back to default code, correct? Still getting the same error in the logs.

And slight correction on serial.log, when changing the log setting in the UI (had been doing it in config.yaml) it does print out a message saying that it's been enabled or disabled, but that's the entirety of the file.

Salandora, 2017-11-04:
Could you please provide your config.yaml, I have a feeling that there is something wrong

akseanor, 2017-11-04:
Sure, it can be found here: https://pastebin.com/RLEPZx8J

I believe I only have added these lines to control a relay via a Python script.:
```
system:
  actions:
  - action: printron
    command: /root/seanor/scripts/printrbotPower.py on
    confirm: false
    name: Printrbot ON
  - action: printroff
    command: /root/seanor/scripts/printrbotPower.py off
    confirm: Turning off the 3D Printer.
    name: Printrbot OFF
```

akseanor, 2017-11-04:
I backed up the config.yaml file on a whim so it would be regenerated from scratch, and it looks like that was indeed the issue. I'm about to attempt to add back in my custom lines to see if those were the problem, but at least now we can close the issue here.

closed 2017-11-04 by akseanor