OctoPrint/OctoPrint #2202: Apache reverse proxy can't login to OctoPrint

LNFWebsite, 2017-11-03:
#### What were you doing?

1. I'm proxying OctoPrint through an Apache reverse proxy per instructions in [Reverse proxy configuration examples](https://github.com/foosel/OctoPrint/wiki/Reverse-proxy-configuration-examples)

2. I'm able to access the OctoPrint interface but unable to login

3. Server returns 400 Bad Request on every login attempt (`POST https://-----------/3d/api/login 400 (Bad Request)`), failing gracefully saying "Login failed / User unknown or wrong password"

Here's my proxying configuration inside `default-ssl.conf`:

		<Location "/3d">
			AuthType Basic
			AuthName "OctoPrint Login"
			AuthBasicProvider file
			AuthUserFile "-------------"
			Require user [username]

			ProxyPass "http://192.168.0.----:5000"
			ProxyPassReverse "http://192.168.0.----:5000"

			RequestHeader set X-SCRIPT-NAME /3d/
			RequestHeader set X-SCHEME https
		</Location>

My preliminary guess is that the issue has to do with the change in protocol from HTTPS into the proxy to HTTP into OctoPrint, but I haven't come up with anyone else who has the same issues when googling...

#### What did you expect to happen?

OctoPrint to login same as when not behind proxy

#### What happened instead?

Login fails gracefully, networks tab in Chrome shows `POST https://-----------/3d/api/login 400 (Bad Request)`

#### Did the same happen when running OctoPrint in safe mode?

Yes.

#### Branch & Commit or Version of OctoPrint

OctoPrint 1.3.4 (master branch)

#### Operating System running OctoPrint

Debian GNU/Linux 8 (jessie)

#### Printer model & used firmware incl. version

Not applicable

#### Browser and Version of Browser, Operating System running Browser

Not applicable, but was latest Chrome on Win8

#### Link to octoprint.log

https://pastebin.com/FxRY8v3U

#### Link to contents of terminal tab or serial.log

https://pastebin.com/FnFhBbGE

#### Link to contents of Javascript console in the browser

https://pastebin.com/UuFWAZ3R

#### Screenshot(s)/video(s) showing the problem:

Not applicable, the login fails saying 'wrong username or password' when inputting correct values

I have read the FAQ.

foosel, 2017-11-03:
I just tested this with your configuration and I cannot reproduce this against 1.3.4, 1.3.5 (which btw is the current version) or 1.3.6.dev:

```
        <Location "/3d">
                AuthType Basic
                AuthName "auth plz"
                AuthUserFile "/etc/apache2/htpasswd/test"
                Require valid-user

                ProxyPass "http://192.168.1.3:5000"
                ProxyPassReverse "http://192.168.1.3:5000"

                RequestHeader set X-SCRIPT-NAME /3d/
                RequestHeader set X-SCHEME https
        </Location>
```

Your logs indicate access via the prefix `/printer` (e.g. `POST https://example.com/printer/api/login 400 (Bad Request)` on your JS console), however your apache config indicates a prefix of `/3d` instead - which of the two now is it and are you sure you are not mixing up two instances here? Your `octoprint.log` shows `405` errors on `/api/login`, which would indicate an attempt to `GET` the request instead of `POST`ing it as defined. I only see two `401` errors that would indicate an invalid username/password combo. I don't see any `400` errors so those don't seem to even reach the instance. Something is not adding up here.

LNFWebsite, 2017-11-03:
Sorry, the discrepancy of `/3d` vs `/printer` is because I copied my Apache config from an earlier question I had asked on a forum. I've been altering the location in trying to fix the bug. I just checked, and the indicated location in the logs is the location listed in my Apache configuration.

I noticed what you said about the 401's. The first of the two 400 requests in the console.log are generated immediately when I open the OctoPrint interface. When attempting to login, another 400 is generated. I'm not sure why OctoPrint would report 401's while I'm given 400's. Maybe OctoPrint is ignoring the 400 to report the 401?

I think the 405's that you see were my attempts to directly access the /api/login directory. I wanted to see the return headers, making sure that this is emanating from the OctoPrint server.

Sorry about that. Also, I'm in the US so sorry for the late replies.

LNFWebsite, 2017-11-04:
Nevermind everything I said... It was a mod_security rule blocking the POST attempt. Thanks for the help anyways!

closed 2017-11-04 by LNFWebsite